bitcraze/crazyflie-lib-python #6: virtualenv

kentwills, 2016-02-26:
'best practice' is usually to install python packages in a virtualenv rather than on the system level, would you accept a pull request that changes the README to provide instructions on installing the cflib package in a virtualenv instead of globally as root?  Furthermore, would you be open to having a Makefile that automates these steps for the user?

ataffanel, 2016-02-26:
Virtualenv would be really nice but there seems to be a problem with PyQT: because pyQT is not in pip it cannot be in a virtualenv. Do you have any clue on that?

kentwills, 2016-02-26:
I just want to make sure that I understand the situation.

To be able to use this package with crazyflie-clients-python, you would not be able to install the library in a virtualenv, because you would have to first ensure that the crazyflie-clients-python project itself could run in a virtualenv.

If I understand the situation correctly, we should probably open a ticket on crazyflie-clients-python to add virtualenv support. (which as noted may be challenging, I can explore more to see what we can do)  We should also provide instructions for installing crazyflie-lib-python globally to be able to use the client.

I think that this library by itself could be used by researchers and developers independently and should not be dependent on the client (the client should be dependent on the lib).  I would recommend showing how to use the examples provided with the lib with a virtualenv (in the way of steps on the README or scripts to support installation).

If you still feel that the only use-case for the lib is going to be the client, leaving the documentation about installing globally wouldn't really hurt things. (if you really must)

ataffanel, 2016-02-26:
You are right, pyqt is a client issue not a lib issue. Then I guess we need the two documentation: virtualenv for the generic case and global/user for running the client.

kentwills, 2016-03-24:
Documentation is mentioned in #10, so closing this as well.

closed 2016-03-24 by kentwills