bitcraze/crazyflie-lib-python #26: Dropping python 2.X?

krichardsson, 2016-12-08:


evoggy, 2016-12-08:
I would suggest keeping the compatibility. I think that a lib should be as compatible/usable as possible and the aim of the Python lib is not only to be used by the client. There's still lots of other libs that's interesting to use together with the Crazyflie but that only supports 2.7+ (like LeapMotion). When implementing the dual support constructs were selected that's supported in both 2.7+ and 3.4+ to minimize using if/else ([see this post](https://www.bitcraze.io/2015/09/moving-to-python-3/))

IBISLabUMD, 2018-06-08:
The question of python2 compatibility is even more important 2 years later. 

Python2 support is being dropped officially on January 1, 2020. Many other packages, including potentially some that folks would use with the crazyflie i.e. numpy, are being dropped sooner. 

It is my belief that any libraries that do not support Python3 at this point indicate that they are outdated or not maintained. Supporting Python2 might be a small burden, but a burden nonetheless.

krichardsson, 2018-08-08:
Thanks for your input!
We will not remove python 2 support for now, but will keep it in mind for the future

krichardsson, 2020-08-05:
We have passed Jan 1 2020 when python 2 is no longer updated (https://www.python.org/doc/sunset-python-2/). We will remove python 2 support from the library.

closed 2020-08-06 by krichardsson