bitcraze/crazyflie-lib-python #47: error when excute the basiclog example

cuongtv89ks, 2017-05-02:
the error:
Scanning interfaces for Crazyflies...
Crazyflies found:
No Crazyflies found, cannot run example
Traceback (most recent call last):
  File "basiclog.py", line 143, in <module>
    while le.is_connected:
NameError: name 'le' is not defined

How to fix it?

thank in advance!!

ataffanel, 2017-05-02:
Did you connect the Crazyflie or a Crazyradio? The examples will connect to the Crazyflie using the Crazyradio dongle.

cuongtv89ks, 2017-05-02:
yes, I plugged Crazyradio dongle and connected to crazyflie but can not run example

cuongtv89ks, 2017-05-02:
oh after restarting again, now it can connect :D

ataffanel, 2017-05-04:
Good to read that the problem is solved. It might have been a problem with access right (after seting up the used rules file, the crazyradio has to be reconnected). I close the ticket.

closed 2017-05-04 by ataffanel