bitcraze/crazyflie-lib-python #49: Tried to run basiclog example: ERROR: No module named examples

shawnarthur, 2017-06-14:
`(venv) Shawn's Mac:crazyflie-lib-python Research$ python -m examples.basiclog
`
I am unable to run this example. Also could you please tell me, how should I edit the source code of the Crazyflie? Is there a tutorial? How to connect to crazyflie? GUI is not showing the radio transmitter platform option.
P.S. I don't have XBOX.

ataffanel, 2017-06-14:
If you have installed the lib with pip, you can just run the examples with the filename: python basiclog.py. This is usually how we run them. What error are you seeing?

For the tutorials there is a getting started tutorial (https://www.bitcraze.io/getting-started-with-the-crazyflie-2-0/) and a getting started with development (https://www.bitcraze.io/getting-started-with-development/) on the Bitcraze website. Following these two documentation will walk you through the process of connecting, flying and programming the Crazyflie.

santolucito, 2017-10-18:
I had the same issue with the error 'crazyflie-lib-python/venv/bin/python2: No module named examples' when using python 2.7. By switching to 3.5 (alias python=python3), I was able to get the basiclog example working (after connecting the crazyflie via usb).

krichardsson, 2017-10-19:
The examples are not designed as modules (running with the -m switch). The intended use is to run

        python examples/basiclog.py

santolucito, 2017-10-19:
Should the README be updated then? atm line 27 of the README tells us to use '-m'

krichardsson, 2017-10-19:
@santolucito yes it should. Sorry, I was a bit too quick on closing this one!
I'll fix it

Thanks!

commit 466334ff (2017-10-19): Updated readme on how to run examples. Closes #49
closed 2017-10-19 by krichardsson