bitcraze/crazyflie-lib-python #58: radio driver : Add a way to configure the value of _RadioDriverThread.TRIES_BEFORE_DISCON

4rzael, 2017-08-18:
Hello,

When working with multiple drones (5) without using broadcast packets, I got drones randomly disconnecting.

This problem was due to packet losses, and the drone being disconnected by [this piece of code](https://github.com/bitcraze/crazyflie-lib-python/blob/master/cflib/crtp/radiodriver.py#L492)

I tried hardcoding `_RadioDriverThread.TRIES_BEFORE_DISCON` to extremely high values, and my problem disapeared, and the drones were still responsive.

It would be great, though, if there was a method call to set this value directly. I did not make a PR as it seems to need some non-trivial changes in order not to be too *hacky*.

ataffanel, 2017-08-24:
Thanks for the ticket, this is a good idea. I am still unsure how to implement that, I have two ideas:

### Global setting in the radio
The number of retry is set globally un the radio driver:
``` python
import cflib.crtp
from cflib.crazyflie import Crazyflie

cflib.crtp.radiodriver.set_retries_before_disconnect(1500)
cflib.crtp.init_drivers()

cf = Crazyflie()
cf.open_link("radio://0/42/2M")
```
Advantage:
 - The setting is for the radio link only, it does not apply to other link
 - One Global setting

Drawback:
 - Cannot be set per Crazyflie connection

### Setting in the Crazyflie
The setting is set in the crazyflie before connection:
``` python
import cflib.crtp
from cflib.crazyflie import Crazyflie

cflib.crtp.init_drivers()

cf = Crazyflie()
cf.set_retries_before_disconnect(1500)
cf.open_link("radio://0/42/2M")
```
Advantage:
 - Expose less 'implementation details' since the setting is generic and does not target the radio
 - Setting can be set for each crazyflie independently

Drawback:
 - Does not make sense for any other link than the radio so having it generic is missleading (ie. this makes not sense for USB link).

4rzael, 2017-08-31:
IMHO, the first way would make more sense. I can't think of any usecases where having a number of retries per crazyflie connection would really matter, and it seems way cleaner to avoid having a generic setting for a specific case.

krichardsson, 2017-12-08:
I implemented option 1 above. If a setting per instance is required in the future, that could be added as well.

closed 2017-12-08 by krichardsson