bitcraze/crazyflie-lib-python #66: Unable to run ramp.py

rafaelrabacca, 2017-12-29:
I'm trying to run the ramp.py file, but I keep encountering an error that I can't understand. 
I'm using windows 10 and a crazyflie 2.0

`Scanning interfaces for Crazyflies...
Crazyflies found:
radio://0/100/2M
ERROR:cflib.crazyflie:Couldn't load link driver: [Errno None] b'libusb0-dll:err [set_configuration] could not set config 1: win error: The parameter is incorrect.\r\n'

Traceback (most recent call last):
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\crazyflie\__init__.py", line 218, in open_link
    link_uri, self._link_quality_cb, self._link_error_cb)
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\crtp\__init__.py", line 91, in get_link_driver
    instance.connect(uri, link_quality_callback, link_error_callback)
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\crtp\radiodriver.py", line 188, in connect
    address)
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\crtp\radiodriver.py", line 98, in __init__
    _RadioManager._radios[self._devid] = _SharedRadio(self._devid)
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\crtp\radiodriver.py", line 67, in __init__
    self.radio = Crazyradio(devid=devid)
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\drivers\crazyradio.py", line 126, in __init__
    self.dev.set_configuration(1)
  File "C:\Users\RAFAEL\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "C:\Users\RAFAEL\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\RAFAEL\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\core.py", line 148, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "C:\Users\RAFAEL\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\backend\libusb0.py", line 493, in set_configuration
    _check(_lib.usb_set_configuration(dev_handle, config_value))
  File "C:\Users\RAFAEL\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\backend\libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [set_configuration] could not set config 1: win error: The parameter is incorrect.\r\n'

Connection to radio://0/100/2M failed: Couldn't load link driver: [Errno None] b'libusb0-dll:err [set_configuration] could not set config 1: win error: The parameter is incorrect.\r\n'

Traceback (most recent call last):
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\crazyflie\__init__.py", line 218, in open_link
    link_uri, self._link_quality_cb, self._link_error_cb)
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\crtp\__init__.py", line 91, in get_link_driver
    instance.connect(uri, link_quality_callback, link_error_callback)
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\crtp\radiodriver.py", line 188, in connect
    address)
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\crtp\radiodriver.py", line 98, in __init__
    _RadioManager._radios[self._devid] = _SharedRadio(self._devid)
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\crtp\radiodriver.py", line 67, in __init__
    self.radio = Crazyradio(devid=devid)
  File "c:\users\rafael\documents\dlsu\thesis\crazyflie\crazyflie-lib-python-master\crazyflie-lib-python-master\cflib\drivers\crazyradio.py", line 126, in __init__
    self.dev.set_configuration(1)
  File "C:\Users\RAFAEL\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "C:\Users\RAFAEL\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\RAFAEL\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\core.py", line 148, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "C:\Users\RAFAEL\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\backend\libusb0.py", line 493, in set_configuration
    _check(_lib.usb_set_configuration(dev_handle, config_value))
  File "C:\Users\RAFAEL\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\backend\libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [set_configuration] could not set config 1: win error: The parameter is incorrect.\r\n'

Connecting to radio://0/100/2M`

ataffanel, 2018-01-02:
Hi, can you please tell a bit more about your setup, what version of python are you using and how did you install it? Also, does the Crazyflie client works properly on your computer when installed from the installer?

rafaelrabacca, 2018-01-04:
I'm trying to run it on the powershell on windows 10
Python version 3.6.4 I installed it using the installer from python.org. It went into my Appdata\local\programs. I had all the optional features ticked (documentation, pip, tcl/tk and IDLE, python test suite, and py launcher)
Crazyflie client is working normally and I used the installer
I've tested the crazyradio on a different computer and it works fine there too.

rafaelrabacca, 2018-01-05:
I switched the crazy radio driver to libusbk instead of libusb-win32. I also switched to python 3.4 and used miniconda to create the environment and let it pick the packages. It seems to be working now.

closed 2018-01-05 by rafaelrabacca

krichardsson, 2018-01-05:
Great!