bitcraze/crazyflie-lib-python #75: How to set_arc for crazyradio using crazyflie class?

barzanisar, 2018-04-02:
Hi,
I am using [cfbridge](https://github.com/dennisss/cfbridge/blob/master/scripts/cfbridge.py) to connect to crazyflie 2.0 from QGC via crazyradio PA. 

Cfbridge creates a crazyflie object that connects to crazyflie using [radiodriver.py](https://github.com/bitcraze/crazyflie-lib-python/blob/master/cflib/crtp/radiodriver.py)

How can I configure the radio's number of retries i.e. in [here](https://github.com/bitcraze/crazyflie-lib-python/blob/master/cflib/crtp/radiodriver.py#L196) from crazyflie class? 

I want to be able set the number of retries in my cfbridge rather than hard coding it in radiodriver.py

ataffanel, 2018-04-04:
There is currently no way to achieve that without modifying the radiodriver code.

I think this case if very similar to the ticket #58 and should be solved the same way so that you could set the number of radio retry as a global option before connecting the Crazyflie.

ataffanel, 2018-04-06:
The PR is merged so I close this ticket. Thanks!

closed 2018-04-06 by ataffanel