bitcraze/crazyflie-lib-python #77: radio-test.py issue

srnetsr, 2018-04-10:
Hi, I am trying to use radio-test.py.
However, when I run it, I got the error message as shown in the following.
```
~/crazyflie-lib-python/examples$ sudo python radio-test.py 
Traceback (most recent call last):
  File "radio-test.py", line 19, in <module>
    import cflib.drivers.crazyradio as crazyradio
ImportError: No module named cflib.drivers.crazyradio
```

Please help me to resolve this issue.

Thank you in advance.

ataffanel, 2018-04-10:
You need to have the lib installed in order to run the script. Either run ```pip install -e .``` from the project root to install the latest development version from source (useful if you plan to work on the lib) or ```pip install cflib``` to install the latest stable version.

ataffanel, 2018-04-16:
It looks like the problem I have been having when testing the script when it arrived in a PR: https://github.com/bitcraze/crazyflie-lib-python/pull/21

You can tweak the script the way I did it or disable bluetooth in the copter (by compiling crazyflie2-nrf-firmware with BLE=0)

srnetsr, 2018-04-17:
Thank you for your reply.
In my case, the address of cf2 was the problem.
The address was set as `0xE7E7E7E701`
After switching it to `0xE7E7E7E7E7`, it worked!

Thank you.

closed 2018-04-25 by srnetsr