bitcraze/crazyflie-lib-python #92: KeyError: 'Variable range.front not in TOC'

Avivm122, 2018-08-21:
Running the multiranger_push.py:
```
/home/aviv-kcg/Desktop/CrazyFlieProject/venv/bin/python /home/aviv-kcg/Desktop/CrazyFlieProject/Projects/multiranger_push.py
Connecting to radio://0/80/250K
Connected to radio://0/80/250K
Traceback (most recent call last):
  File "/home/aviv-kcg/Desktop/CrazyFlieProject/Projects/multiranger_push.py", line 79, in <module>
    with Multiranger(scf) as multiranger:
  File "/home/aviv-kcg/crazyflie-lib-python/cflib/utils/multiranger.py", line 111, in __enter__
    self.start()
  File "/home/aviv-kcg/crazyflie-lib-python/cflib/utils/multiranger.py", line 64, in start
    self._cf.log.add_config(self._log_config)
  File "/home/aviv-kcg/crazyflie-lib-python/cflib/crazyflie/log.py", line 413, in add_config
    raise KeyError('Variable {} not in TOC'.format(name))
KeyError: 'Variable range.front not in TOC'

Process finished with exit code 1
```

I'm getting this error, anyone knows why?

ataffanel, 2018-08-21:
This error is most likely accurate: your Crazyflie does not have the required log variable. What version of the firmware are you running? I have tried with the latest version and it works (I do not have a multiranger deck installed so the demo starts the Crazyflie and lands it right away).

Avivm122, 2018-08-21:
How can i check?

ataffanel, 2018-08-21:
You can check the build by lookins at the log tab in the Crazyflie client after connecting the Crazyflie.

The best to make sure you are running the latest version is to build and flash the crazyflie firmware by yourself from bitcraze/crazyflie-firmware. Since the multiranger deck is not released yet, you need to compile the firmware from git to get a firmware that supports it.

Avivm122, 2018-08-21:
I see a lot of values, which one represent the firmware version?

ataffanel, 2018-08-21:
Sorry I did a typo earlier, you can check the version in the console tab.

In the log TOC tab you can check if the variables you want to use exists.

Avivm122, 2018-08-21:
```
SYS: Build 0:6ee723ad1144 (2018.01.1) MODIFIED
SYS: I am 0x313139313535511435003B and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_DRIVERS: Found 11 drivers
DECK_INFO: Found 1 deck memory.
DECK_CORE: 1 deck enumerated
Motion chip is: 0x49
si pihc noitoM: 0xB6
MPU9250 I2C connection [OK].
AK8963 I2C connection [OK].
LPS25H I2C connection [OK].
ESTIMATOR: Using estimator 2
EEPROM: I2C connection [OK].
AK8963: Self test [OK].
DECK_CORE: Deck 0 test [OK].
SYS: Free heap: 16056 bytes
```

Thats what im getting

ataffanel, 2018-08-21:
This is older than the commit that sets the current multiranger variable name(https://github.com/bitcraze/crazyflie-firmware/commit/e9e84fd92eafb62789c402fb228c1fe30c0d7b6b). The solution is to compile and flash a firmware from the git source code.

Just to check: do you have a multiranger deck already? If not, this example will not be useful to you anyway.

Avivm122, 2018-08-21:
yes i do.
Ive downloaded and compiled the firmware but i cant realize how to flash it.

ataffanel, 2018-08-21:
```make cload``` and booting the Crazyflie in bootloader mode is what you need to do to flash the firmware.

ataffanel, 2018-08-27:
Since this does not seems to be a bug in the lib I am closing the issue.

closed 2018-08-27 by ataffanel