bitcraze/crazyflie-lib-python #98: Bootloader is sometime not flashing reliably

ataffanel, 2018-09-05:
There has been more occurrence of the Bootloader flashing a corrupt firmware which prevents the Crazyflie from starting and requires to flash the firmware again. The bootloader protocol is designed to be safe so there is a strong assumption that something has been broken in the python bootloader client implementation.

ataffanel, 2018-09-05:
It has been found experimentally that when the corruption occurs, a full 10 pages of flash is kept un-programmed (ie. all 0xff). The commit ed9d093 flushes the downlink before attempting to flash 10 pages, this should make sure that we are not mistaken a previous flash ack as the current flash ack and should guarantee that all pages are written.

ntamas, 2019-12-17:
We are also experiencing occasional flashing issues on our side; I managed to catch one occasion now and in our case, the first 1600 bytes of a full 10-page buffer were repeated from the previous full 10-page chunk (basically, the range [0x1e000; 0x1e640) was repeated from [0x1b800; 0x0x1be40) ). This happened with the up-to-date master of the Crazyflie client, but we are using the command line to flash the new firmware, not the GUI. Could this be related? (It might not be a coincidence that 1600 bytes divided by 25 (which is the number of bytes that we can update in the upload buffer with a single CRTP packet) is 64, a power of 2).

I was looking into the code of the command-line firmware updater, and I noticed that there are no response packets for the `UPDATE_BUFFER` requests that fill the next set of 10 pages into the buffer that is then written to the flash. How can the client be sure that the packets containing the updated data reached the upload buffer?

ntamas, 2019-12-17:
After some more debugging (and printing packets to stdout), here is my conjecture. What _probably_ happens is as follows:

1. The client manages to fill the buffer successfully and send a request to the STM32 to flash the upload buffer to the flash memory.

2. Flashing takes more than one second for some reason, and in the meanwhile, the client considers the request packet to be lost, and re-sends the flashing request packet.

3. Flashing finishes on the STM32 so it puts a confirmation packet in the outbound queue and then resumes processing incoming requests. (If I read the STM32 bootloader code correctly, it does not process inbound packets while it is flashing).

4. The STM32 sees the _new_ flashing request packet so it starts flashing the same page _again_.

5. In the meanwhile, the client sees the confirmation for the _first_ flashing request from the STM32, which it mistakenly thinks that it is for the _second_ request (as it considers the first request to be lost), so it starts sending requests to update the upload buffer to contain the next page from the firmware.

6. However, note that the STM32 is still busy flashing the previous page for the _second_ time, so it does not process the inbound buffer update packets. I'm not familiar with the system enough to figure out what happens to these packets -- maybe the nRF51 confirms them as received, but the STM32 drops them?

7. The STM32 finishes flashing the previous page and sends the confirmation packet to the client. However, the client does not care about it (it already thinks that the second flashing request has already finished on the STM32), and happily keeps on sending the buffer update packets. _If_ the STM32 ignored the buffer update packets while it was still flashing, it _can_ happen that the next flashing request will flash a buffer where some initial prefix of the buffer is not updated.

I have added a `print("Re-sending flash request")` statement to `cloader.py` at the appropriate place for sake of testing, and then flashed the same firmware to the same Crazyflie ten times. Eight times out of those ten attempts, there was no printout about re-sent flash requests, and the flashing process went fine. In the remaining two times, there _was_ one re-sent flash request, and the Crazyflie failed to boot after the flashing, so I'm pretty sure there's a connection between re-sent flash requests and the firmware corruption during upload.

@ataffanel, does that sound reasonable? If so, is there any workaround for it other than reading the entire buffer back before sending a flashing request? (I'm currently experimenting with it and will get back to you with more updates as soon as I have something to report).

ataffanel, 2019-12-17:
This does match very well our observation so far, I was suspecting something like that but was not able to observe it.

Increasing the timeout for waiting for the flashing to happen might help. I am really sad I did not add sequence numbers or page number to the flash response, it would have allowed to see when that happens.

We have seen the problem happens more when the firmware image is bigger. The page size in the STM32F405 are of different size and from ~128K we hit 128KB pages:
[image]

According to the datasheet, 128K pages takes more than one seconds to erase
[image]

We definitly need to increase the timeout for the programming step, at least for the boundaries where the firmware will erase a page (the bootloader communicates a table that describes the flash architecture to the client).

ntamas, 2019-12-17:
> The page size in the STM32F405 are of different size and from ~128K we hit 128KB pages
That explains a _lot_ - when the re-sending happened, it was always around 128K or so.

FYI, increasing the timeout of flashing request packets from 1 seconds to 2 seconds seemed to have resolved the problem - I managed to flash a firmware successfully 10 out of 10 times. (Did not test further).

One thing that we can also try is as follows: _if_ a flashing request was indeed re-sent, we could try comparing the first few bytes of the upload buffer with what should be there. If the two are different, there was an error while flashing so we need to re-flash the last page, probably with a timeout that is larger than what it was before. In my experience, the difference is always at the beginning of a 10-page boundary, and if the first few bytes are correct, the chances are high that the rest of the page is also correct.

ataffanel, 2019-12-17:
This sounds great, thanks a lot for the investigation and test!

I think the increased timeout should definitely be pushed-in (maybe 2.5 seconds is safer since it is above the max possible erase time).

Do you plan to push a PR or shall I do the change?

ntamas, 2019-12-17:
I will send a PR later today when I get back to my desk.

T.

ataffanel, 2020-01-27:
After more test, this issue has not been seen again. The fix seems to be working great, thanks @ntamas !

I am then closing this issue for good :).

commit 08fa149a (2019-12-18): increase write flash request timeout to 2.5s, fixes #98
commit 5bd21fcc (2019-12-18): Merge pull request #133 from collmot/bugfix/fix-incorrect-flashing-on-stm32

Increase write flash request timeout to 2.5s, fixes #98
closed 2020-01-27 by ataffanel