bitcraze/crazyflie-lib-python #105: Ambiguous README description

hanzheteng, 2019-01-21:
The following part is now clear to me.
```
Linux, OSX, Windows

    Build a virtualenv (local python environment) with package dependencies
        pip install virtualenv
        virtualenv venv
        source venv/bin/activate
    pip install -r requirements.txt
    Activate the environment: source venv/bin/activate
    Connect the crazyflie and run an example: python examples/basiclog
    Deactivate the virtualenv if you activated it deactivate

Note: For systems that support make, the first four steps can be replaced with make venv

Note: The first three steps can be skipped if you don't mind installing cflib dependencies system-wide.
```

What do you mean by the `the first four steps` and `the first three steps`? I am not sure which part is necessary and which is not. I think I did this part both inside and outside `virtualenv` environment. Also, in which working directory should I execute the above commands? Since you mentioned `make env`, I guess it should be done within `crazyflie-lib-python`, but not sure if it's the same to `virtualenv`. 

On the other hand, what's the influence if I install it system-wide? Recently I have gotten in trouble with my python environment and I am not sure if `virtualenv` has changed some configuration in my `~/.local` directory, since I run it in home directory `~`.

krichardsson, 2019-02-09:
I tried to clean up the readme, please close if it is good enough. If not, feedback or a pull request is welcome!

The problem with installing dependencies system wide is that you might have other python code that has the same dependency but requires another version. Cflib only depends on pyusb>=1.0.0b2 at the moment.

hanzheteng, 2019-02-09:
A short question:

If I install dependencies within `virtualenv`, then every time when I want to launch `cfclient`, I have to activate `virtualenv` first right? After using `cfclient` I can then `deactivate` it.

krichardsson, 2019-02-11:
I'm not using virtual environments my self, but that is my understanding.

closed 2019-02-12 by hanzheteng