bitcraze/crazyflie-lib-python #108: Unable to retrieve complete pose data on Roadrunner using python library.

ashwinsushil, 2019-02-16:
Hello everyone,

This is kind of a weird issue that I'm facing.

I'm trying to communicate with a Roadrunner using the crazyflie-lib-python to retrieve the pose of the Roadrunner. I'm using the 'autonomousSequence.py' in the examples with the quaternions added like below,

```
def position_callback(timestamp, data, logconf):
    x = data['kalman.stateX']
    y = data['kalman.stateY']
    z = data['kalman.stateZ']

    q0 = data['kalman.q0']
    q1 = data['kalman.q1']
    q2 = data['kalman.q2']
    q3 = data['kalman.q3']

    print('pos: ({}, {}, {}, {}, {}, {}, {})'.format(x, y, z, q0, q1, q2, q3))

def start_position_printing(scf):
    log_conf = LogConfig(name='Position', period_in_ms=500)
    log_conf.add_variable('kalman.stateX', 'float')
    log_conf.add_variable('kalman.stateY', 'float')
    log_conf.add_variable('kalman.stateZ', 'float')

    log_conf.add_variable('kalman.q1', 'float')
    log_conf.add_variable('kalman.q2', 'float')
    log_conf.add_variable('kalman.q3', 'float')
    log_conf.add_variable('kalman.q0', 'float')

    scf.cf.log.add_config(log_conf)
    log_conf.data_received_cb.add_callback(position_callback)
    log_conf.start()
```
Unfortunately I'm not able to get the pose when all these variables are added.
And the output is as below
```
Connecting to usb://0
Connected to usb://0
Waiting for estimator to find position...

```
 **However when less than 7 variables are added, for example like below. I'm able to get the data** 

```
def position_callback(timestamp, data, logconf):
    x = data['kalman.stateX']
    y = data['kalman.stateY']
    z = data['kalman.stateZ']

    q0 = data['kalman.q0']
    q1 = data['kalman.q1']
    q2 = data['kalman.q2']
    # q3 = data['kalman.q3']

    print('pos: ({}, {}, {}, {}, {}, {})'.format(x, y, z, q0, q1, q2))

def start_position_printing(scf):
    log_conf = LogConfig(name='Position', period_in_ms=500)
    log_conf.add_variable('kalman.stateX', 'float')
    log_conf.add_variable('kalman.stateY', 'float')
    log_conf.add_variable('kalman.stateZ', 'float')

    log_conf.add_variable('kalman.q0', 'float')
    log_conf.add_variable('kalman.q1', 'float')
    log_conf.add_variable('kalman.q2', 'float')
    # log_conf.add_variable('kalman.q3', 'float')
    
    scf.cf.log.add_config(log_conf)
    log_conf.data_received_cb.add_callback(position_callback)
    log_conf.start()
```
**OR**
```
def position_callback(timestamp, data, logconf):
    x = data['kalman.stateX']
    y = data['kalman.stateY']
    z = data['kalman.stateZ']

    # q0 = data['kalman.q0']
    q1 = data['kalman.q1']
    q2 = data['kalman.q2']
    q3 = data['kalman.q3']

    print('pos: ({}, {}, {}, {}, {}, {})'.format(x, y, z, q1, q2, q3))

def start_position_printing(scf):
    log_conf = LogConfig(name='Position', period_in_ms=500)
    log_conf.add_variable('kalman.stateX', 'float')
    log_conf.add_variable('kalman.stateY', 'float')
    log_conf.add_variable('kalman.stateZ', 'float')

    # log_conf.add_variable('kalman.q0', 'float')
    log_conf.add_variable('kalman.q1', 'float')
    log_conf.add_variable('kalman.q2', 'float')
    log_conf.add_variable('kalman.q3', 'float')

    scf.cf.log.add_config(log_conf)
    log_conf.data_received_cb.add_callback(position_callback)
    log_conf.start()
```
Does anyone have any idea whats happening ? 
To be clear my goal is to retrive x,y,z,q0,q1,q2,q3.

**Firmware version on Roadrunner:** 
SYS: Build 4:1762deae7f25 (2019.02 +4) CLEAN
SYS: I am 0x3934323530374718003F0023 and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_INFO: DECK_FORCE=bcDWM1000 found
DECK_INFO: compile-time forced driver bcDWM1000 added
DECK_CORE: 1 deck(s) found
DECK_CORE: Calling INIT on driver bcDWM1000 for deck 0
IMU: BMI088 Gyro I2C connection [OK].
IMU: BMI088 Accel I2C connection [OK]
ESTIMATOR: Using Kalman (2) estimator
CONTROLLER: Using PID (1) controller
EEPROM: I2C connection [OK].
DECK_CORE: Deck 0 test [OK].
SYS: Free heap: 11432 bytes
TDOA3: 2D positioning enabled at 0.300000 m height
DWM: Automatic mode: detected TDoA3

**LPS node firmware**: lps-node-firmware-2018.10.dfu

**System details:**
Ubuntu 16.04
ROS Kinetic

Thanks in advance,
Ash

krichardsson, 2019-02-18:
Hi!

This is actually by design. There is a buffer in the Crazyradio that sets an upper limit to the size of the log data that fits into one packet. In the python lib this is mirrored in the LogConfig class that holds one log block, unfortunately there is no good feedback that the block is over sized. However it is possible to have multiple log blocks so just add a more LogConfigs and all should be good.

The limitation is due to the HW implementation of current Crazyradio and it is likely that this will change in the future with larger buffers. We have discussed adding a higher level logging class in the python lib to handle this property and hide the limitation to the user, but we have not got around to do it yet.

ashwinsushil, 2019-02-18:
@krichardsson  Thank you mate. That makes it clear. :)

closed 2019-02-18 by ashwinsushil