bitcraze/crazyflie-lib-python #109: Fail to get logs if we restart the script

AlexisTM, 2019-02-18:
*Bug explanation:*

The logging system seems unreliable when trying to reconnect to the Crazyflie (Roadrunner) after it was soft disconnected.

**Expected behaviour:**

* Run `basiclog.py`
* Have data printed
* `basiclog.py` stops after 5 seconds
* Run `basiclog.py` directly after
* Have data printed
* `basiclog.py` stops after 5 seconds

**Experienced behaviour:**

* Run `basiclog.py`
* Have data printed
* `basiclog.py` stops after 5 seconds
* Run `basiclog.py` directly after
* *Nothing is printed and *
* `basiclog.py` stops after 5 seconds
* *OR*
* `basiclog.py hangs on 'Connecting to usb://0`

AlexisTM, 2019-02-18:
Second run of `basiclog.py` with **3 variables**

```
Scanning interfaces for Crazyflies...
Crazyflies found:
usb://0
Connecting to usb://0
Connected to usb://0
Disconnected from usb://0
```

Second run of `basiclog.py` with **6 variables**, it freezes.

```
Scanning interfaces for Crazyflies...
Crazyflies found:
usb://0
Connecting to usb://0
```

krichardsson, 2019-02-18:
I can reproduce this using USB. Seems to work with a radio. 
This is most likely a firmware problem, I created an issue for it bitcraze/crazyflie-firmware#394

AlexisTM, 2019-03-18:
Very likely to be due to the STM USB library: https://github.com/bitcraze/crazyflie-firmware/issues/393

closed 2019-03-18 by AlexisTM