bitcraze/crazyflie-lib-python #110: Roadrunner (maybe CF) goes in unstable state if the node is stopped with Ctrl+Z

AlexisTM, 2019-02-18:
**Bug explanation:**

When we stop a script which uses logging with CTRL+Z, the roadrunner seems to enter an inconsistent state. Killing the script afterwards with `kill -9 %1` does not solve the issue on the roadrunner.

**Reproduce:**

- Start `examples/basiclog.py`
- Stop the script with CTRL+Z
- The LEDs stops blinking (except one red LED of the RX?) 

https://www.youtube.com/watch?v=W0XmP-DjeLs&feature=youtu.be

krichardsson, 2019-02-18:
Hi!

I did try to reproduce the problem and at first I could not make it crash. By unplugging the radio and use USB I did manage to get a similar behaviour though. Were you using USB or a radio?

There is definitely a problem in the firmware, I created an issue for it bitcraze/crazyflie-firmware#393.

By the way, the video is not public, I can not watch it.

AlexisTM, 2019-02-18:
I am using the USB interface.

I get randomly the same behaviour using the crazyflie_driver node for ROS.

Note: The video has been switched to unlisted. 

https://www.youtube.com/watch?v=W0XmP-DjeLs

krichardsson, 2019-02-18:
Yes, the video is consistent with bitcraze/crazyflie-firmware#393.

The five fast blinks of the red LED is indicating that there was an error during start up, either self test failed or that it was restarted after a crash. 

When I connect to it with the client after the crash, the log in the console tab contains:
```
SYS: Roadrunner 1.0 is up and running!
SYS: Build 0:a1d5aac6f3e6 (2019.02) MODIFIED
SYS: I am 0x373336383436471700280033 and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_DRIVERS: Found 16 drivers
DECK_INFO: Found 0 deck memory.
DECK_INFO: compile-time forced driver bcDWM1000 added
DECK_CORE: 1 deck enumerated
IMU: BMI088 Gyro I2C connection [OK].
IMU: BMI088 Accel I2C connection [OK]
ESTIMATOR: Using estimator 2
CONTROLLER: Using controller 1
EEPROM: I2C connection [OK].
DECK_CORE: Deck 0 test [OK].
SYS: The system resumed after watchdog timeout [WARNING]
SYS: No assert information found
```

AlexisTM, 2019-03-18:
This is likely a Firmware bug due to STM USB library: https://github.com/bitcraze/crazyflie-firmware/issues/393

closed 2019-03-18 by AlexisTM