bitcraze/crazyflie-lib-python #112: high-level position commander crashes with similar setpoints

knmcguire, 2019-03-15:
The high-level commander seems to crash if the two setpoints given in sequence are the same. Here is a minimal example code for the python script:

 ```
with PositionHlCommander(
            scf,
            x=0.0, y=0.0, z=0.0,
            default_velocity=0.3,
            default_height=1.0,
            controller=PositionHlCommander.CONTROLLER_PID) as pc:
        
        time.sleep(1.0)
        pc.go_to(0.0,0.0,0.5)
        time.sleep(1.0)
        pc.go_to(0.0,0.0,0.5)
       
       # Here the CF drops from the sky
```

If the setpoints are different, this does not happen.

knmcguire, 2019-03-29:
This issue should have been fixed at crazyflie_firmware level with commit: https://github.com/bitcraze/crazyflie-firmware/commit/db7864651adf0062e4005435e9451d0b0d0180ef. It is not test flown yet but I guess we can close this issue now?

ataffanel, 2019-04-01:
Well, I have been thinking about is and it still does not seem right for the client to send a goto command with 0 as time. It is very good that it is fixed in the Crazyflie but it feels that the lib needs to be fixed as well.

One way to fix it would be to modify the position commander to not send a goto command if the requested position is the same as the current position. I cannot see any negative effect  of doing so and it will prevent the lib from sending a goto with 0 as time.

closed 2019-08-13 by krichardsson