bitcraze/crazyflie-lib-python #122: Reading one-wire memories via USB during initialization hangs sometimes

ntamas, 2019-07-18:
I am experiencing issues with connecting to a Crazyflie with the Python library via USB. The Crazyflie has a Loco positioning deck on the top and a LED ring on the bottom; both of which provide a one-wire memory that appears at IDs 5 and 6 when querying the memories. Connecting to the Crazyflie via USB often hangs; the link is established and the library starts downloading the TOCs and the memory information, but the "connected" event is never emitted.

I have tracked the problem down to the point where the one-wire memories are read during the initialization process. It seems like read requests are sent simultaneously to both memories, and the request sent to memory ID 5 (the one corresponding to the LED ring) often fails in the sense that the response callback for the read request is not called. Since the USB connection is deemed reliable by the library (i.e. no resending is needed), the request is never re-sent and the initialization process hangs.

I have patched the initialization process such that it reads the one-wire memories sequentially instead of in parallel and it seemed to have resolved the problem, although I'm not sure this is the way to go. Is anyone else experiencing this problem?

ntamas, 2019-07-18:
Forgot to mention that the problem does not appear if I try to connect to the Crazyflie via the radio; I guess this is because the radio connection automatically re-sends packets when needed.

krichardsson, 2019-10-21:
Thanks for the information, I have not seen this problem but we have made some updates of the USB parts recently and maybe there is a connection. Which firmware version did you see this in?

ntamas, 2019-10-21:
I haven't seen this problem recently either with fresh git checkouts and a freshly compiled firmware. I guess we can close this for the time being; I'll re-open if it emerges again.

closed 2019-10-21 by ntamas