bitcraze/crazyflie-lib-python #126: How to get the distance between Anchor and Cf(deck) in LPS?

Supredan, 2019-09-04:
Hi,

I want to get the distance between Anchor and Cf(deck) in LPS,and i found [lps-ros](https://github.com/bitcraze/lps-ros) which is exactly what i need .but unfortunately I have encountered the difficulties described [here](https://forum.bitcraze.io/viewtopic.php?t=3240).which says "The problem comes from the fact that this doc is very outdated and should not be followed" And suggested that I use crazyfile-ros and cflib,So I want to know how to get the distance data via cflib and send it using the ros node.

Thank!

krichardsson, 2019-09-11:
Please see [this answer in the forum](https://forum.bitcraze.io/viewtopic.php?f=3&t=3229&p=17780&hilit=tdoa+ros#p17780)

The log system is sampling the variable in the Crazylie and you will probably not get all TDoA data this way. The log system is unfortunately not really designed for this type of tasks. 

If you want all data it is possible to add a streaming protocol, [something similar to this](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/crtp_localization_service.c#L219
), but it requires some work in the firmware and modifications of the ROS driver. 

Other users have streamed TDoA data through a UART to another device, but then you will have a cable connected that will prevent flying :-)

Supredan, 2019-09-14:
I will continue this topic in the forum, let us close him here

closed 2019-09-14 by Supredan