bitcraze/crazyflie-lib-python #130: Radio close() slow on VMWare with USB3

ataffanel, 2019-11-28:
According to this forum thread: https://forum.bitcraze.io/viewtopic.php?f=12&t=4029 there is a long delay when closing the link if the lib is used in VMWare with USB3 enabled.

The problem seems to be located around the usb reset() call, and replacing reset() by ```usb.util.dispose_resources(self.dev)``` in the crazyradio driver seems to fix the problem.

Calling a USB reset was done as a precaution but should not be absolutly required and is known to be incompatible with some libusb backend (lilke winusb) so it would be a good thing for other platform anyway to remove all call to usb reset.

ntamas, 2019-12-17:
This happens not only on VMWare but also on a standard MacBook Pro.

ataffanel, 2019-12-27:
@ntamas Thanks, this confirms that we should avoid using USB reset everywhere we can.

commit acdcfd00 (2020-01-07): USB: remove calls to USB reset(), fixes #130
closed 2020-01-07 by ataffanel