bitcraze/crazyflie-lib-python #131: RSSI from lib-python

PartiallyObservable, 2019-12-06:
Hi,

   I have several crazyflies and crazy radio PA's. I am trying to set up the crazyswarm software and part of that is assigning channels to all of the radios/crazyflies, so I am attempting to perform some radio tests to see which channels work best in my lab.

The example script "crazyflie-lib-python/examples/radio-test.py" seems to do exactly what I need it to (measure ack rate and rssi on all the channels), but for some reason it is not returning an rssi value for any channel even though the ack rates seem good. 

Specifically in this portion of the code (line 73)
```
if pk.ack and len(pk.data) > 2 and \
           pk.data[0] & 0xf3 == 0xf3 and pk.data[1] == 0x01:
            # append rssi data
            temp.append(pk.data[2])
```
It seems that len(pk.data) is almost always 1. Oddly right after I boot the crazyflie it seems that it does have len(pk.data) > 2, but only for a few seconds (and I am unsure if the actual data is garbage or is actually an rssi) 

Is this script outdated? According to this [commit](https://github.com/bitcraze/crazyflie2-nrf-firmware/commit/0649c56a7d08bcd90bbbd80f18f900b261bda80a) it seems like there should be rssi data available at data[2], but I might be interpreting this wrong. Thank you for any help!

ataffanel, 2019-12-09:
I just tried and you need to add a compile flag to the nrf firmware in order to enable RSSI measurement.

If you clone https://github.com/bitcraze/crazyflie2-nrf-firmware you can compile and bootload the firmware with:
``` bash
$ make clean
$ make BLE=0 CONFIG=-DRSSI_ACK_PACKET
$ make BLE=0 CONFIG=-DRSSI_ACK_PACKET cload
```

If that works for you please confirm here, I think we need to update the script to add a warning about it.

I am not sure why the RSSI is not enabled by default though, I think that could be something nice to have all the time.

PartiallyObservable, 2019-12-11:
Thank you for your help. Adding that compile flag fixed my problem and I now receive RSSI information which is helpful.

I have an additional question about [Auto ACK configuration](https://wiki.bitcraze.io/doc:crazyradio:usb:index#auto_ack_configuration). In the linked documentation it mentions 

> By default, the Crazyradio is configured with auto ACK enabled. This means that after transmitting a packet, the radio waits for an acknowledge from the receiver

If this is the case, should the acknowledgement rate calculated by radio-test.py always be equal to 1? This is not the behaviour I am seeing (For most channels it is 1, but some are lower).  Thanks again.

ataffanel, 2019-12-11:
Great, thanks for the update. I will update the script with more information.

For the auto ack, Crazyradio sends a packet and wait for an Ack from the Crazyflie. The ack packet may or may not be received, what the script is measuring is the ratio packet sent/ack received.

There can be at least two reasons why an ack is not received: either the packet send has been lost, or the ack has been lost. In both case it correspond to packet loss, most likely due to collision with another radio, and this is what this test is measuring.

closed 2019-12-11 by ataffanel

swarm5, 2020-07-24:
> I just tried and you need to add a compile flag to the nrf firmware in order to enable RSSI measurement.
> 
> If you clone https://github.com/bitcraze/crazyflie2-nrf-firmware you can compile and bootload the firmware with:
> 
> ```shell
> $ make clean
> $ make BLE=0 CONFIG=-DRSSI_ACK_PACKET
> $ make BLE=0 CONFIG=-DRSSI_ACK_PACKET cload
> ```
> 
> If that works for you please confirm here, I think we need to update the script to add a warning about it.
> 
> I am not sure why the RSSI is not enabled by default though, I think that could be something nice to have all the time.
@ataffanel 
1. I don’t know how to upload the compiled firmware. I don’t know what I did, right?
When I use "`make BLE=0 CONFIG=-DRSSI_ACK_PACKET cload`", the terminal has the following prompt:

> python3 -m cfloader flash cf2_nrf.bin nrf51-fw
> Restart the Crazyflie you want to bootload in the next
>  10 seconds ..

I tried restarting cf, but it didn't work. Finally, I entered bootloader mode and it successfully uploaded the firmware. The terminal has the following prompt:

> done!
> Connected to bootloader on Crazyflie 2.0 (version=0x10)
> Target info: nrf51 (0xFE)
> Flash pages: 232 | Page size: 1024 | Buffer pages: 1 | Start page: 88
> 144 KBytes of flash available for firmware image.
> Target info: stm32 (0xFF)
> Flash pages: 1024 | Page size: 1024 | Buffer pages: 10 | Start page: 16
> 1008 KBytes of flash available for firmware image.
> Flashing 1 of 1 to nrf51 (fw): 66071 bytes (65 pages) .1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1 .1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1
> Reset in firmware mode ...

2. When I run` radio-test.py`, I encountered a problem that some or all of the RSSI data is missing.
When I run radio-test.py, only part of the time, all data points of ACC and RSSI are available, and more often, part or all of the RSSI data is missing. The following is a typical situation when I run radio-test.py:

> ('Channel', 69, 'ack_rate:', 1, 'rssi average:', 69.170000000000002, 'rssi std:', 0.375632799419859)
> ('Channel', 70, 'ack_rate:', 1, 'rssi average:', 70.870000000000005, 'rssi std:', 0.33630343441600474)
> ('Channel', 71, 'ack_rate:', 1, 'rssi average:', 72.640000000000001, 'rssi std:', 0.52000000000000002)
> ('Channel', 72, 'ack_rate:', 0, 'rssi average:', 76.525773195876283, 'rssi std:', 0.71928921824793446)
> ('Channel', 73, 'ack_rate:', 0, 'rssi average:', 85.13333333333334, 'rssi std:', 0.80553639823963807)
> /usr/lib/python2.7/dist-packages/numpy/core/_methods.py:59: RuntimeWarning: Mean of empty slice.
>   warnings.warn("Mean of empty slice.", RuntimeWarning)
> /usr/lib/python2.7/dist-packages/numpy/core/_methods.py:70: RuntimeWarning: invalid value encountered in double_scalars
>   ret = ret.dtype.type(ret / rcount)
> /usr/lib/python2.7/dist-packages/numpy/core/_methods.py:82: RuntimeWarning: Degrees of freedom <= 0 for slice
>   warnings.warn("Degrees of freedom <= 0 for slice", RuntimeWarning)
> /usr/lib/python2.7/dist-packages/numpy/core/_methods.py:94: RuntimeWarning: invalid value encountered in true_divide
>   arrmean, rcount, out=arrmean, casting='unsafe', subok=False)
> /usr/lib/python2.7/dist-packages/numpy/core/_methods.py:116: RuntimeWarning: invalid value encountered in double_scalars
>   ret = ret.dtype.type(ret / rcount)
> ('Channel', 74, 'ack_rate:', 0, 'rssi average:', nan, 'rssi std:', nan)
> ('Channel', 75, 'ack_rate:', 0, 'rssi average:', nan, 'rssi std:', nan)
> ('Channel', 76, 'ack_rate:', 0, 'rssi average:', nan, 'rssi std:', nan)
> ('Channel', 77, 'ack_rate:', 0, 'rssi average:', nan, 'rssi std:', nan)

[image]
[image]
[image]

swarm5, 2020-07-24:
@ataffanel @whoenig 
While unable to accept the RSSI value, I tried to make cload again. The terminal reports an error: usb.core.USBError: [Errno 16] Resource busy.
When I re-plugged the USB crazyradio PA, it can successfully make cload.
The firmware of my crazyradio PA uses the version provided by crazyswarm. I don't know if it will affect it.

swarm5, 2020-07-24:
I did not find a problem similar to mine. I would like to try another computer to determine if this problem is caused by my computer.

whoenig, 2020-07-25:
Please open a new issue rather than posting on closed issues. Closed issues do not show up in the github dashboard, so it's easy to be missed and never get any response on. I haven't tested that script recently, but it looks like you might have run into some regression bug here.

commit 87a6769a (2019-12-11): Closes #131: Update radio-test.py with info about required nrf build flags