bitcraze/crazyflie-lib-python #144: cflib.crtp.scan_interfaces() crashes on OSX

krichardsson, 2020-03-24:
Pull request #140 introduced [scanning for serial ports](https://github.com/bitcraze/crazyflie-lib-python/blob/0.1.10/cflib/crtp/serialdriver.py#L147-L152) when scanning for interfaces. 

```
    def scan_interface(self, address):
        if found_serial:
            devices_names = [x.name for x in list_ports.comports()]
            return [('serial://' + x, '') for x in devices_names]
        else:
            return []
```

The code assumes the device has a name, which does not seem to be the case on OSX, where name is None. This causes the code to throw an exception and terminate the scanning.

Found on:
* OSX 10.15.3
* Python 3.7.3
* python-lib 0.1.10
* pyserial 3.4

One possible solution could be to use the ```device``` member of the device instead of the ```name``` member. It seems to be populated, on OSX at least.

krichardsson, 2020-03-25:
Hi @s9meboeh and @Darneas!

I can see that you had a discussion with @knmcguire in #140 about the URIs for serial connections. It has turned out that the solution does not work on OSX and I'm not looking for a fix to this issue. 

The root cause seems to be that "name" is None for the ports on my machine. I see two solutions
1. Skip all ports where name==None - might lead to ports not being accessible
1. Use the "device" member instead - URIs will look like ```serial:///dev/cu.URT0```

Another issue (slightly off topic but I'll bundle it in this communication anyway) is that the serial ports show up in the Crazyflie python client as well, which is probably not that useful to most users. My suggestion is to add a flag (or call) to the library that must be used to enable the serial port support. This would keep the client clean without modifying the API.

Any input is welcome!

s9meboeh, 2020-03-26:
Hello,
I like the form with the short name because it is easy to read and has no triple slashes.
What do you think of enableling both versions as URI (`serial://<name>` and `serial://<device>`)?
Or do you think that causes to much confusion?

For the serial ports not to show up in the Python client it would be an option to add another optional parameter to `cflib.crtp.init_drivers` as there is already for the debug driver.

krichardsson, 2020-03-27:
@s9meboeh  thanks for the feedback. 
I have implemented the changes, could you try it out on your side? I don't have any device to test on.

s9meboeh, 2020-03-30:
Hello,
I tested your changes with our setup and it works perfectly fine.

closed 2020-03-30 by krichardsson