bitcraze/crazyflie-lib-python #146: Add an script for easy tuning of the PID controller

knmcguire, 2020-03-30:
This script will make it easier to tune the PID controller in a crazyflie. It contains a graph to see the effect of the step response and the gains sent. This is all tested with a lighthouse system.

- [ ] Position Tuning
- [ ] Velocity Tuning
- [ ] Attitude Tuning
- [ ] Tuning with a flowdeck

fabrizioschiano, 2020-04-28:
Hi @knmcguire , thanks for this script. I was wondering if:
- you could add the PID values corresponding to the different lines in the plot above.  
- you have an idea on why there seems to be a "delay" in the first part of the curves (more visible in the red and light-red curves) 

Thanks!

knmcguire, 2020-04-29:
The gains of the newest line can be found in the screenshot. The old gains are found in the firmware (https://github.com/bitcraze/crazyflie-firmware/blob/5c5448f303ab9b419b72649d22546eb9b28cf0ac/src/modules/src/position_controller_pid.c#L108)

That delay has a very simple explanation :) The whole PID control structure is heavily cascaded (see [this documentation](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/controllers/)n for the control structure. The delay means that the innerloops of the PID controller are not tuned optimally. That is why we need this script to work for velocity, attitude and attitude rate as well, but that is difficult to do in a controlled environment as it is with the position. 

I am curious though how this would look if the new INDI controller is used for attitude (Outerloop should then still be PID). I will give it a go somewhere in these weeks.

fabrizioschiano, 2020-04-30:
Thanks, I also thought about that and I think it makes sense. 
@knmcguire, are you thinking about building a small test rig to tune the attitude and attitude rate controllers?

knmcguire, 2020-04-30:
There was also an speed limitation that was set to 0.3 m/s (since I'm using the highlevel commander for the script). Increasing this to 1 m/s results in the following plot:

[image]

At the end it seems faster, but it does not explain the slow startup. I will try to increase the gains of the velocity controller. Apparently, it is the case that the position controller feed the desired velocity to the velocity controller before going to the attitude controller, so Ill need to update the documentation. 

@fabrizioschiano Currently we are not planning to build a rig, but it might be the best way to tune this properly indeed. Ill discuss this with the rest if we want to work on that this quarter.

knmcguire, 2020-05-11:
Btw, I tried the INDI controller as the attitude controller but the delay still persists. I would need to make a velocity  version of this controller script, since the position controller actually feeds command to the velocity controller directly (currently in the documentation this is said that it is interchangeable but actually after investigation, this is not the case). 

Once tune the velocity controller with the new script, hopefully that will fix the delay up to a certain point. Anydelay after that, needs to be done with the attitude controller, which needs this special testrig we talked about.

fabrizioschiano, 2020-05-11:
@knmcguire, thanks for the update. Fingers crossed that the velocity controller tuning will be enough (but I doubt it).

knmcguire, 2020-05-28:
Tuning the velocity controller first does not make a difference in reducing this delay (0.25 sec in z and even 0.4 sec in xy). Mellinger controller is more able to meet this step response but it flips quite a lot. Maybe that is also an indication that the requirements to meet this step response exactly are too much for the motors. 

It is also better not to use the high level commander for generating the step response as it more of smooth trajectory for both position and velocity and some angles as well. It would be good to have a script that really seperates these responses more sharply.

HazarBoughanmi, 2021-08-28:
Hey ,I'm working with Loco  Positioning  system LPS and when I run this script it show me some errors 
any idea please?

Connecting to radio://0/80/1M/E7E7E7E7E7
Connected to radio://0/80/1M/E7E7E7E7E7
Waiting for estimator to find position...
Traceback (most recent call last):
  File "PID_controller_tuner.py", line 395, in <module>
    cf_ctrl = TunerControlCF(pid_gui, cf)
  File "PID_controller_tuner.py", line 165, in __init__
    self.cf.param.request_param_update('posCtlPid.zKp')
  File "/home/bitcraze/projects/crazyflie-lib-python/cflib/crazyflie/param.py", line 247, in request_param_update
    self.param_updater.request_param_update(
  File "/home/bitcraze/projects/crazyflie-lib-python/cflib/crazyflie/param.py", line 341, in request_param_update
    pk.data = struct.pack('<B', var_id)
struct.error: required argument is not an integer

knmcguire, 2021-08-31:
Hi! Do you know which version of the cflib you are using? You can use:

`pip freeze | grep cflib`

And is the crazyflie firmware updated to the latest release?

aminbassiri, 2022-09-06:
How can I run the script, what's the command? 
how to run the PID tuner crazyflie? I mean with which command?

knmcguire, 2022-09-09:
The script is in examples/tuning/ folder in this repo

However I would strongly advise you to use the new PID tuning tab of cfclient instead, described here: https://www.bitcraze.io/2022/08/cfclient-gui-lighthouse-and-bolt-improvements/