bitcraze/crazyflie-lib-python #147: P2P API lib-python

pablogs9, 2020-04-14:
Hello!

I was wondering if there any chance of using the new P2P API with a python script in order to send messages between a second Crazyflie Radio (different of the one controlling the drone) and the drone.

In a more general term: is there any possibility of connecting a Crazyflie to a two Crazyflie Radios? One for controlling and the second for exchanging data for example?

Thanks!

knmcguire, 2020-04-15:
Yes that is absolutely possible. I did it with the out-of-tree build (app-layer) in here: https://github.com/tudelft/SGBA_CF2_App_layer. In this case, both CR's were using p2p, but you can also have one communicating with your CFclient and the other using p2p.  Do be aware that all need to be on the same channel and datarate!

pablogs9, 2020-04-16:
Thanks really useful, I have read that it is required to have the NRF51 firmware built without BT support: [here](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/p2p_api/). 

Are there somewhere some detailed instructions for building and flashing this NRF51 chip?

And, I have seen that your Python code sends P2P packets to the CF, but there is any way to receive P2P commands from CF in the Python scripts?

knmcguire, 2020-04-16:
actually, with the recent master of the firmware of nrf51, it is no longer necessary to build without BT support. But I advise to still to do so just to be sure

How to build and flash the nrf firmware can be found here: https://github.com/bitcraze/crazyflie2-nrf-firmware.

I would say yes, but a quick look at the cflib says that there is not an only-listen functionality. There is however within the send packet an ack possibility, so that it sends a packet and waits for the answer. That is pretty much how the regular communication is handled.

pablogs9, 2020-04-20:
Please, let me know if I'm wrong with any of these assumptions:

- I can flash bootloader (downloaded a built from master here: https://github.com/bitcraze/crazyflie2-stm-bootloader) by using DFU mode and dfu-util at address `0x08000000`
- I can flash firmware (downloaded a built from master here: https://github.com/bitcraze/crazyflie-firmware) by using DFU mode and dfu-util at address `0x08004000`
- Then I should be able to enter in bootloader mode by connecting USB and pressing CF2.1 button while the LED is flashing slow (just before entering en DFU).
- Then I should be able to flash NRF firmware (downloaded and built from master here: https://github.com/bitcraze/crazyflie2-nrf-firmware) by using `cfloader` and a CrazyRadio.

Well, I have been able to reproduce these steps just once. Then I'm not able to enter en bootloader mode anymore.

Do you think that not having the battery connected can be an issue?

knmcguire, 2020-04-20:
There were issues with the crazyflie not going into bootloader mode if the battery was not connected in some version of the crazyflie firmware... So do try with a battery connected.

Hmm... I wouldn't touch the bootloader or dfu upload at all if you have a crazyradio. You might break it and then you would need to get a debugger adapter to fix it. Just use 'make cload' in the nrf repo if you want to flash the latest software (I know it is not in the documentation but we will fix it soon.)

By the way, since it is now moving in a different topic, we use the issue list to post about issues with the code or improvements. General support questions we prefer to have on forum.bitcraze.io. So if you have anymore questions, could you please post there so that I can close this issue?

pablogs9, 2020-04-20:
Of course I'll moving to the forum, thanks.

closed 2020-04-20 by pablogs9