bitcraze/crazyflie-lib-python #148: Establishing communication with simulated cf instead of real cf

EnricaSo, 2020-04-24:
I am trying to use the [`PID_controller_tuner`](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/tuning/PID_controller_tuner.py) to tune a simulated crazyflie rather than a real one. For the simulation, I'm using a [Github project](https://github.com/wuwushrek/sim_cf) that allows performing SITL simulation of the crazyflie in Gazebo. 
To connect to it from the controller example, I changed the URI [here](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/tuning/PID_controller_tuner.py#L45) to ```URI = 'udp://127.0.0.1:19950'``` since I know the address and port to which the simulated crazyflie is communicating. 
However, I get this exception:
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/esoria/Developer/crazyflie-lib-python/cflib/crazyflie/__init__.py", line 386, in run
    pk = self.cf.link.receive_packet(1)
  File "/home/esoria/Developer/crazyflie-lib-python/cflib/crtp/udpdriver.py", line 65, in receive_packet
    data, addr = self.socket.recvfrom(1024)
ConnectionRefusedError: [Errno 111] Connection refused

Connecting to udp://127.0.0.1:19950
```
Do you know what it could be due? Am I using the wrong approach?

Many thanks!

krichardsson, 2020-04-28:
I don't know if the approach is good or not, but the error (ConnectionRefusedError) looks like the lib can not connect to the port. Are you sure it is the correct port and that the library is allowed to access it?

EnricaSo, 2020-04-30:
Thank you for your answer. I continued to investigate and found the following.
The URI of the crazyflie was not correct, the result of `netstat` indeed is:
```
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name   
udp      768      0 0.0.0.0:49439           0.0.0.0:*                           14880/cf2       
udp        0      0 0.0.0.0:19950           0.0.0.0:*                           12423/gzserver
```
Now, when I change the uri to the one of the simulated crazyflie [here](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/tuning/PID_controller_tuner.py#L45) like this:
```
# URI = 'radio://0/80/2M/E7E7E7E7E7'
URI = 'udp://127.0.0.1:49439'
```
the application [PID_controller_tuner](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/tuning/PID_controller_tuner.py) throws this error:
```
Connecting to udp://127.0.0.1:49439
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/esoria/Developer/crazyflie-lib-python/cflib/crazyflie/__init__.py", line 386, in run
    pk = self.cf.link.receive_packet(1)
  File "/home/esoria/Developer/crazyflie-lib-python/cflib/crtp/udpdriver.py", line 65, in receive_packet
    data, addr = self.socket.recvfrom(1024)
ConnectionRefusedError: [Errno 111] Connection refused
```
and `tcpdump` outputs this:
```
sudo tcpdump -i lo "udp and (not port 19950)"
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on lo, link-type EN10MB (Ethernet), capture size 262144 bytes
11:49:50.342425 IP localhost.40115 > localhost.7777: UDP, length 5
11:49:50.342685 IP localhost.40115 > localhost.7777: UDP, length 3
```

From my understanding, PID_controller_tuner is trying to communicate with `7777` instead of `49439` (the port of the cf-firmware).
Therefore, I replace `7777` with `49439` in the [udpdriver](https://github.com/bitcraze/crazyflie-lib-python/blob/e515becb64967d527ca6bb729ad6c501695f20d0/cflib/crtp/udpdriver.py#L58), but now I have a different error:
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/esoria/Developer/crazyflie-lib-python/cflib/crazyflie/__init__.py", line 386, in run
    pk = self.cf.link.receive_packet(1)
  File "/home/esoria/Developer/crazyflie-lib-python/cflib/crtp/udpdriver.py", line 71, in receive_packet
    pk.data = data[1:]
  File "/home/esoria/Developer/crazyflie-lib-python/cflib/crtp/crtpstack.py", line 129, in _set_data
    self._data = bytearray(data)
ValueError: byte must be in range(0, 256)

Connecting to udp://127.0.0.1:49439
```

Do you see anything wrong? I am not an expert in communication, but connecting the PID_controller_tuner to the simulated crazyflie would be great for tuning the crazyflie in simulation.

Best regards,
Enrica

krichardsson, 2020-05-05:
Comment part 1:

It actually looks like the UDP driver ignores the address/port in the URI and always uses localhost/7777. I'm not aware of the history of this driver, but it is possible that it was implemented for a specific use case. Anyway it should be fairly simple to parse the URI and set the address and port here https://github.com/bitcraze/crazyflie-lib-python/blob/0.1.10/cflib/crtp/udpdriver.py#L58.

Please consider contributing if you fix it :-)
I added a new issue #150

krichardsson, 2020-05-05:
Comment part 2:

You seem to have got past the connection issue above and the error you get (ValueError: byte must be in range(0, 256)) is probably caused by the unpacking of data in https://github.com/bitcraze/crazyflie-lib-python/blob/0.1.10/cflib/crtp/udpdriver.py#L65-L72

The code is fairly old and probably written for python2, it is possible that something has changed for python3?

santiagorg2401, 2020-07-29:
Hi, how did you get the crazfie port?

EnricaSo, 2020-07-30:
Hi @santiagorg2401, how are you simulating your drones? In my case, I was using [sim_cf](https://github.com/wuwushrek/sim_cf). Once you have generated the Crazyflie instances you can check their address with `netstat`.

santiagorg2401, 2020-07-30:
Hi,
I followed the guide on the project you used, using SITL simulation
In one terminal window I launch gazebo 
`roslaunch crazyflie_gazebo crazyflie_sim.launch `
In other I execute this:
`./run_cfs.sh `
And I click the play button on Gazebo.
When I run netstat (In other terminal) I can't find the address and port of the simulated Crazyflie

jonasdn, 2021-06-08:
Replaced with #240

closed 2021-06-08 by jonasdn