bitcraze/crazyflie-lib-python #151: Add an example of fusing external position/pose information from vicon_bridge into crazyflie

JIMMMY0, 2020-05-08:
Assume we have obtained position or pose of the crazyflie through [vicon_bridge](https://github.com/ethz-asl/vicon_bridge), how to fuse these external information into crazyflie?  I think this is a high-demanding function for those who use VICON system. 
Really appreciate if an example could be provided. In this way, we don't need to immerse the whole system into ROS (except for vicon_bridge) as in [crazyflie_ros](https://github.com/whoenig/crazyflie_ros).

krichardsson, 2020-05-08:
Hi!

I agree, this would be a nice example. Unfortunately we don't have a vicon system but we would be happy to accept a pull request if someone else wrote it.

There is a similar [example for a Qualisys system](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/qualisys/qualisys_hl_commander.py) that can be used as a starting point.

JIMMMY0, 2020-05-12:
Thanks for your suggestion. We used `vicon_bridge` to get the drone's pose and wrote a ROS node trying to fuse the pose information into drone's kalman filter by using the method  `Extpos.send_extpos(self, x, y, z)`. And then reset the kalman filter as in the qualisys example. One question is do we have to call the function `wait_for_position_estimator(cf)` as in [qualisys example](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/qualisys/qualisys_hl_commander.py) ? To put it short, **assume we have already obtained the external position of cf, how to correctly fuse it into Kalman filter and reset it with initial position** ？ We would appreciate if a minimum example could be provided.

Another problem is if we use the crazy radio to coneect to the drone for sending external pose information, if we send control commands(such as velocity command) at the same time, it will show that the crazyradio is busy. Any suggestion on this ? I think this issues is quite related to the first problem. If we can correctly send external pose to the drone, sending control commands will not be a problem.

krichardsson, 2020-08-05:
Hi, sorry for the very late reply!
The call to `wait_for_position_estimator(cf)` is to make sure the kalman filter has converged to the correct pose. Feeding data through `Extpos.send_extpos(self, x, y, z)` uses a very low std and the filter should converge quickly so I guess it is not really necessary with a mocap system.

> if we send control commands(such as velocity command) at the same time, it will show that the crazyradio is busy

How often do you send the pose data? Could it be that you saturate the channel? You probably don't have to send the pose data very often. At some point I remember pushing pose data at only 5 Hz, but it probably depends on your application.

closed 2021-09-06 by jonasdn

knmcguire, 2023-08-10:
@abdul-mannan-khan we don't have a minimal example since we don't have an vicon system but I'll make a seperate issue for this. 

For other support, please use our [discussions](https://discussions.bitcraze.io/)