bitcraze/crazyflie-lib-python #152: Make more send_setpoint functions

knmcguire, 2020-05-11:
Currently there is no way to send a setpoint from the library for attitude rate, which has been requested from this issue  from the firmware: https://github.com/bitcraze/crazyflie-firmware/issues/572. This used to be possible but that was before the current commander framework was implemented a few years ago.

jonasdn, 2021-09-06:
@knmcguire do you have any tips for anyone looking to implement this?

knmcguire, 2021-09-06:
The [flightmode.stabmoderoll/pitch](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/api/params/#flightmode) seems to be needed to put to one in order to make the crazyflie react to the attitude setpoints in a rate PID manner.

knmcguire, 2024-04-24:
So this is already possible if this function is used:

`send_setpoint(self, roll, pitch, yawrate, thrust)`

while the `flightmode.stabmodroll/pitch` are set to 1

I'll add this to the documentation so that this is clear to the users that this is possible.

We also heard from some people that they'll like to set the motors directly but that would require starting a new port and a setpoint structure for the rpm/volt per motor, so that should be a new issue.

knmcguire, 2024-04-24:
made new issue in firmware to address making an crtp commander port to set motors https://github.com/bitcraze/crazyflie-firmware/issues/1372

PR #452 (2024-05-07): mention rate mode setpoint set in doc
fixes #152
closed 2024-05-07 by knmcguire