bitcraze/crazyflie-lib-python #160: crazyflie is not stable

gavtech, 2020-07-28:
Hello,
I am trying to fly crazyflie but it is not flying properly, it is going sideways and it is very unstable. I loaded the firmware and there is no flow deck. Sometimes its running better and sometimes it is not.  
I would like to ask is flow deck necessary to make it stable?
And may I know what is the problem for this.

Thank you

krichardsson, 2020-07-28:
Hi!
You need some sort of positioning support (like the Flow deck), the IMU is too noisy for stable autonomous flight for more than a very short time.

The best medium for this type of question is our forum (https://forum.bitcraze.io/), we try to use github for code related discussions. Thanks!

closed 2020-07-28 by krichardsson

gavtech, 2020-07-28:
Okay thank you.