bitcraze/crazyflie-lib-python #166: Qualisys example script sensitive to errors when pushing full pose

krichardsson, 2020-09-02:
The https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/qualisys/qualisys_hl_commander.py script gets the full pose from the QTM and pushes it into the Crazyflie using the send_extpose() function of the Extpos class in the python lib. For unknown reasons, the script seems to be sensitive to errors in the pose, resulting in a CF that flips or go in the wrong direction. It very much looks like the estimated position in the CF is wrong.

There are lots of possible error sources:
* Unexpected data from the Qualisys system
* A bug in the script when pushing the pose
* The std used when pushing the pose into the EKF might be off
* A bug in the firmware handling pose data
* Delays

This requires more investigations.

Using only the position works fine (through the send_extpos() method).

knmcguire, 2022-03-07:
#305 was perhaps related to this, but let's see.

It would perhaps be good to implement this python library of libmotion capture instead as that is more often tested?

https://github.com/IMRCLab/libmotioncapture

I mentioned this in the cfclient issue: https://github.com/bitcraze/crazyflie-clients-python/issues/577

knmcguire, 2022-09-06:
In Crazyswarm(2) the variance of the external position is adjusted by standard before pushing the mocap results to the EKF. So we need to do that in this script as well.

PR #359 (2022-09-07): Fix Qualisys example
The qualisys example has not been fully functional when using full pose. This PR fixes this.

Closes #166
closed 2022-09-07 by evoggy