bitcraze/crazyflie-lib-python #172: scan_interfaces() no longer works

EceChaik, 2020-10-25:
Hi everyone, 

after changing the address of my crazyflie to 60/2M/E7E7E7E7AF
the basilog.py example, which uses scan_interfaces() , can no longer find the crazyflie. It says no crazyflies found.
I know the crazyradio and the crazyflie work, because when I use the crazyflie_ros package I can connect to the crazyflie on its new uri, no problem.
So why could it be that scan_interfaces no longer finds the crazyflie? Isn't it supposed to look at all addresses?

krichardsson, 2020-10-26:
Hi!

The scan searches all channels, but only for the configured address (0xE7E7E7E7E7 by default).
You can set the address in the scan with ```scan_interfaces(address=0xE7E7E7E7AF)```

EceChaik, 2020-10-28:
Thanks, feel free to close

closed 2020-10-28 by krichardsson