bitcraze/crazyflie-lib-python #181: isAlive() deprecated; switch to is_alive()

ManCla, 2020-12-03:
in the crazyflie class, open_link() function, line 232
there is a check that the thread is alive but the function name has been changed the definition of the thread class

the issue appeared both when trying to connect to the crazyflie through the client and when running some of the examples from the library (e.g. ramp.py)

 - I am using python3

thecountoftuscany, 2020-12-07:
Facing the same issue. Traceback below:
```
Connecting to radio://0/13/2M/E7E7E7E7E7
ERROR:cflib.crazyflie:Couldn't load link driver: '_IncomingPacketHandler' object has no attribute 'isAlive'

Traceback (most recent call last):
  File "/another/path/crazyflie-lib-python/cflib/crazyflie/__init__.py", line 234, in open_link
    if not self.incoming.isAlive():
AttributeError: '_IncomingPacketHandler' object has no attribute 'isAlive'

Connection to radio://0/13/2M/E7E7E7E7E7 failed: Couldn't load link driver: '_IncomingPacketHandler' object has no attribute 'isAlive'

Traceback (most recent call last):
  File "/another/path/crazyflie-lib-python/cflib/crazyflie/__init__.py", line 234, in open_link
    if not self.incoming.isAlive():
AttributeError: '_IncomingPacketHandler' object has no attribute 'isAlive'

Traceback (most recent call last):
  File "/some/path/cflibController.py", line 203, in <module>
    with SyncCrazyflie(args.uri, cf=Crazyflie(rw_cache=os.path.expanduser("~") + "/.cache")) as scf:
  File "/another/path/crazyflie-lib-python/cflib/crazyflie/syncCrazyflie.py", line 72, in __enter__
    self.open_link()
  File "/another/path/crazyflie-lib-python/cflib/crazyflie/syncCrazyflie.py", line 69, in open_link
    raise Exception(self._error_message)
Exception: Couldn't load link driver: '_IncomingPacketHandler' object has no attribute 'isAlive'

Traceback (most recent call last):
  File "/another/path/crazyflie-lib-python/cflib/crazyflie/__init__.py", line 234, in open_link
    if not self.incoming.isAlive():
AttributeError: '_IncomingPacketHandler' object has no attribute 'isAlive'

```

ataffanel, 2020-12-08:
Thank you for the report, what version of python are you using, is it 3.9?

ManCla, 2020-12-08:
Yes!

closed 2020-12-10 by ataffanel

ataffanel, 2020-12-10:
Interistingly this bug did not show of when using the client (the only thing I had tested on 3.9 so far). I pushed a fix that solved the issue and allows to run the examples.

ManCla, 2020-12-10:
That weird, I attach a screenshot of how the bug appeared to me. I am also sure it is using python 3.9.0 (inserted a print statement to check it).

Anyhow it looks good now, thanks for your work!

[image]

ataffanel, 2020-12-10:
I had not installed the lib from source when running the client ... maybe the bug was inserted after the last release then. Anyway, good that it now works :-).

As a side note since you are on Mac: it does work on Big Sur as well now :-).

mbaytas, 2021-01-25:
@ataffanel -- This problem persists, throwing an `AttributeError` in **/examples/qualisys/qualisys_hl_commander.py** line 303, with Python 3.9 on Windows:

    Exception has occurred: Exception
    Couldn't load link driver: '_IncomingPacketHandler' object has no attribute 'isAlive'

    Traceback (most recent call last):
      File "C:\Users\mbaytas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\cflib\crazyflie\__init__.py", line 232, in open_link
        if not self.incoming.isAlive():
    AttributeError: '_IncomingPacketHandler' object has no attribute 'isAlive'
      File "C:\Users\mbaytas\Code\qualisys_hl_commander.py", line 303, in <module>
        with SyncCrazyflie(uri, cf=Crazyflie(rw_cache='./cache')) as scf:

ataffanel, 2021-02-08:
@mbaytas The fix was not part of a release yet. We have now published a new version on PyPi.

commit f7704d77 (2020-12-10): Closes #181: Replace isAlive() with is_alive()