bitcraze/crazyflie-lib-python #183: Connected callback not being invoked

jancoow, 2020-12-17:
Hi,

I have noticed a bug when using the crazyflie 2.1 with the crazyflie-lib. 
I was interested in using the radiobridge to send mavlink messages over syslink. However, I could never get the example to work because there were no messages received in qgroundcontrol.
I figured out that the callback "connected" in the CrazyFlie() class is never invoked. Why is this? 

By using the callback link_established instead it seems to work great.

ataffanel, 2020-12-17:
Hi,

Thanks for the report and the quick PR!

I have never used this bridge, it was contributed, though the line you are changing should work as are as I understand. This callback is used in most examples, for example in [ramp.py](https://github.com/bitcraze/crazyflie-lib-python/blob/f7704d77ab8d712fdd92531bb6d33b36d3965233/examples/ramp.py#L49), and is called when the connection process is complete.

The link_established callback is called much earlier in the connection process and things like the log and param subsystem will not be initialized at that point.

Can you try to run the ramp.py example and see if it works? There might be a problem elsewhere. More often than none, connection problems comes from running a recent lib against an old firmware so upgrading the Crazyflie firmware to the latest release might help.

I will keep the PR on hold until we have figured out what is happening.

ataffanel, 2020-12-17:
Hi,
I was thinking about this issue, and since the point of this script is to interface PX4, the connection will likely never succeed since it assumes the stock firmware is running in the Crazyflie.

This script should not use the Crazyflie object at all, it should open the link and use it directly. I can accept your pull request since it seems to solve the problem (likely by making the connection process hang). An improvement would be to use the link instead of the Crazyflie object in the future.

closed 2020-12-17 by ataffanel