bitcraze/crazyflie-lib-python #185: ValueError exception in qualisys_hl_commander.py

mbaytas, 2021-01-20:
With Python 3.9 on Windows, line 116 `async for qtm_instance in qtm.Discover('0.0.0.0'):` raises the following exception and the script does not run:

    Exception has occurred: ValueError
    Passing `reuse_address=True` is no longer supported, as the usage of SO_REUSEPORT in UDP poses a significant security concern.
      File "C:\Users\baytas\Desktop\qualisys_hl_commander.py", line 116, in _discover
        async for qtm_instance in qtm.Discover('127.0.0.1'):
      File "C:\Users\baytas\Desktop\qualisys_hl_commander.py", line 101, in _connect
        qtm_instance = await self._discover()
      File "C:\Users\baytas\Desktop\qualisys_hl_commander.py", line 95, in _life_cycle
        await self._connect()
      File "C:\Users\baytas\Desktop\qualisys_hl_commander.py", line 92, in run
        asyncio.run(self._life_cycle())

krichardsson, 2021-01-20:
As far as I understand this is a problem in the Qualisys lib https://github.com/qualisys/qualisys_python_sdk/issues/13

mbaytas, 2021-01-25:
I was able to bypass the issue by modifying the `_connect` function:

    async def _connect(self):
            # qtm_instance = await self._discover()
            # host = qtm_instance.host
            host = "127.0.0.1"

closed 2021-01-25 by mbaytas