bitcraze/crazyflie-lib-python #196: Add support for crazyflie-link-cpp

whoenig, 2021-02-19:
We developed a new link library in C++ for lower latency, higher bandwidth, and lower CPU utilization in https://github.com/bitcraze/crazyflie-link-cpp. This library already comes with Python bindings to be used with Python 3.x.

We first need to add support in crazyflie-lib-python, that can be enabled on demand using environment variable `USE_CFLINK=cpp`. Once we can confirm that it works well on all target platforms, we can switch to this backend by default.

jonasdn, 2021-03-17:
What actions are left before this can be closed?

whoenig, 2021-03-17:
This hasn't been tested thoroughly on all platforms yet. Ultimately, the goal was to enable this backend by default and possibly remove the older pure Python implementation in order to minimize combinations that we have to test.

jonasdn, 2021-03-17:
How do we reach the level of testing we need? Can we put-out a call for testing? Do we have go-to persons? Or should we just do it ourselves? Are there areas you are more worried about than others?

whoenig, 2021-03-17:
We already know that there are crashes on shutdown on Windows in the lib, so this needs to be fixed first. Otherwise the normal release-cycle testing will suffice (including our automated system tests, which we only ran from a Linux host so far).

ntamas, 2021-03-25:
Did you experience any crashes on shutdown on other platforms? In particular, I use an async-await-based alternative Crazyflie library and I was trying to make it work with `crazyflie-link-cpp` as it would simplify things a great deal when crossing from coroutine-world to thread-world and back (as the threading part would nicely be hidden in the C++ layer). I work on macOS and I get crashes like this upon exiting the Python interpreter:

```
Assertion failed: (pthread_mutex_destroy(mutex) == 0), function usbi_mutex_destroy, file ./os/threads_posix.h, line 58.
```

It comes from the `libusb_exit()` call and somehow it seems not to be consistent; I cannot reproduce it with simple scripts, only with a more complex test case that imports lots of stuff from my library in order to do a throughput test. I was just wondering whether this assertion failure is the same as the one you see on Windows.

The full stack trace of the crashed thread is as follows:

```
Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff68a5a33a __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff68b16e60 pthread_kill + 430
2   libsystem_c.dylib             	0x00007fff689e1808 abort + 120
3   libsystem_c.dylib             	0x00007fff689e0ac6 __assert_rtn + 314
4   cflinkcpp.cpython-39-darwin.so	0x000000010b31a92f usbi_io_exit.cold.5 + 33
5   cflinkcpp.cpython-39-darwin.so	0x000000010b313bb7 usbi_io_exit + 160
6   cflinkcpp.cpython-39-darwin.so	0x000000010b3115ff libusb_exit + 440
7   cflinkcpp.cpython-39-darwin.so	0x000000010b2bbbbc bitcraze::crazyflieLinkCpp::USBManager::~USBManager() + 28 (USBManager.cpp:83)
8   cflinkcpp.cpython-39-darwin.so	0x000000010b2bbca5 bitcraze::crazyflieLinkCpp::USBManager::~USBManager() + 21 (USBManager.cpp:84)
9   libsystem_c.dylib             	0x00007fff689bc13c __cxa_finalize_ranges + 319
10  libsystem_c.dylib             	0x00007fff689bc412 exit + 55
11  libdyld.dylib                 	0x00007fff68912cd0 start + 8
```

jonasdn, 2021-03-25:
What version of libusb do you have? I got these with 1.0.24 but not 1.0.23.

ntamas, 2021-03-25:
I have 1.0.24; I'll try 1.0.23 later (it's hard to install older versions from Homebrew so I would need to compile it on my own). Thanks for the tip!

ntamas, 2021-03-25:
Okay, it turns out that the problem was that I was doing something like this in my main async loop to avoid blocking the loop when I was about to close the connection:

```
await run_sync_in_worker_thread(connection.close)
```

Apparently this call never returned, i.e. `connection.close()` was never executed on the worker thread (I have to investigate why), so when `libusb_exit()` was called the connection was still open. If I call `connection.close()` on the main async loop then it works just fine -- it could theoretically block the main loop, though, so I need to investigate whether this could be a problem, but that's a whole different story.

PR #233 (2021-05-10): Set cflinkcpp as default link driver on x86_64
- Radiodriver: Set cflinkcpp as default
- setup.py: Require cflinkcpp on x86_64

Closes: #196
closed 2021-05-10 by jonasdn