bitcraze/crazyflie-lib-python #202: Documentation for `send_extpose`

alvinsunyixiao, 2021-02-28:
I am trying to integrate with the Vicon system in our lab. Following the `qualisys` example, I successfully achieved position tracking with `crazyflie.extpose.send_extpos` (without orientation measurement). However, when I tried to send the orientation via `send_extpose(x, y, z, qx, qy, qz, qw)` API, the drone just went crazy. **What coordinate system is that quaternion living in?** The forum has a discussion thread around this problem [here](https://forum.bitcraze.io/viewtopic.php?t=4279), but there isn't a good resolution other than not sending the orientation info to the drone.

I also noticed some confusing logic in the qualisys example
1. In [this](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/qualisys/qualisys_hl_commander.py#L220) function, the conversion from rotation matrix to quaternion is strange. Why only the diagonal entries of the rotation matrix is involved here? The conversional formula for the conversion is not like this. So what are the meanings of those equations?
2. [Here](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/qualisys/qualisys_hl_commander.py#L135) the rotation matrix is column major. Is it transposed (inverse)? What direction of the rotation is it describing? From world to drone or from drone to world?

whoenig, 2021-02-28:
The send_extpose functions takes the orientation in the world frame, so you should be able to directly include the values as reported from the VICON DataStream SDK. However, make sure that you define the rigid body in Vicon Tracker such that positive x is front (where the antenna is), and positive y is on the left side of the drone. If you still have problems, try adjusting the kalman filter, similar what is done in the Crazyswarm: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L29-L33.

For your other questions, I believe that this is related to poor orientation tracking in the Bitcraze test lab, where 'axis flips' frequently occur, which make the roll and pitch measurements not useful. The method in the code seems to include yaw only. If you need general conversion functions, I can recommend [rowan](https://rowan.readthedocs.io/en/latest/package-rowan.html).

alvinsunyixiao, 2021-02-28:
Thanks a lot. Recalibrating with the correct rigid body frame fixes this issue.

closed 2021-02-28 by alvinsunyixiao