bitcraze/crazyflie-lib-python #210: Client crashing when exiting

ataffanel, 2021-03-11:
The client is often crashing when exit. On Linux is used to segfault, it apparently now segfaults on Mac and on my machine it exists succesfully with the message:
```
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.8/logging/__init__.py", line 2115, in shutdown
    h.close()
RuntimeError: wrapped C/C++ object of type LogClientTab has been deleted
```

I am running: 
```
Cfclient version: 2021.1-43-g4fa2a8b
System: linux
Python: 3.8.6
Qt: 5.15.2
PyQt: 5.15.3
```

jonasdn, 2021-03-18:
I think this crash is hiding an abort now.

The `LogClientTab` adds itself as a Python logging loggandler. And it receives events even after the QT widget has been deleted. If I remove the logging that goes on at sigint I instead get an abort:

```
python3: os/threads_posix.h:46: usbi_mutex_lock: Assertion `pthread_mutex_lock(mutex) == 0' failed.
Aborted (core dumped)
```

jonasdn, 2021-03-18:
Downgrading libusb from 1.0.24 to 1.0.23 (`sudo dnf downgrade libusbx`) made the pthread assert go away for me.

jonasdn, 2021-03-18:
The LogClientTab was fixable by not using a Qt widget class as a log handler.

closed 2021-03-23 by ataffanel