bitcraze/crazyflie-lib-python #240: get value error in connection with sim_cf

poritech, 2021-05-28:
Hello!

I want to control a simulated crazyflie with [sim_cf](https://github.com/wuwushrek/sim_cf).
I have successfully connected by referring to #148, but I get the error `ValueError: byte must be in range(0, 256)`.

### connection procedure
1. create the environment by referring to the readme of [sim_cf](https://github.com/wuwushrek/sim_cf).
2. Launch gazebo and the simulator
```
cd ~/catkin_ws/src/sim_cf/crazyflie_gazebo/scripts
roslaunch crazyflie_gazebo crazyflie_sim.launch
./run_cfs.sh
```
3. connect
Use the function `SyncCrazyflie` in crazyflie-lib-python to connect. Set uri to `udp://0.0.0.0:55502` for example. The port can be referenced during simulator startup with `netstat -p -a | grep cf`.
`udp        0      0 0.0.0.0:55502           0.0.0.0:*                           4998/cf2`

### error output details
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/tyoneda/anaconda3/envs/crazyfile/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/tyoneda/Documents/bitcraze/crazyflie-lib-python/cflib/crazyflie/__init__.py", line 392, in run
    pk = self.cf.link.receive_packet(1)
  File "/home/tyoneda/Documents/bitcraze/crazyflie-lib-python/cflib/crtp/udpdriver.py", line 79, in receive_packet
    pk.data = data[1:]
  File "/home/tyoneda/Documents/bitcraze/crazyflie-lib-python/cflib/crtp/crtpstack.py", line 128, in _set_data
    self._data = bytearray(data)
ValueError: byte must be in range(0, 256)
```

This error was mentioned in #148, but since the issue was out of date, I created a new issue.

jonasdn, 2021-06-08:
Hi @poritech !

The referenced PR fixes the ValueError (among other things), are you able to get further with this?

Regards
Jonas

poritech, 2021-06-11:
Hi jonasdn!

Sorry for the late reply.
And thank you for bug fix!

Value Error is no longer detected, but the program is stuck at SyncCrazyflie() and can not execute next step.
Are the values returned by the simulator different from those of Crazyflie?

I don't know if this will help, but here this is a dump of the data received by udpdriver from the simlator.
```
1,144,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
1,82,1,180,14,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,221,250,173
1,82,0,182,14,1,184,153,66,186,186,23,112,186,53,211,0,61,191,102,155,190,252,9,171,190,221,250,173
1,82,1,188,14,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,221,250,173
```

jonasdn, 2021-06-11:
Hi, no worries!

There has been some confusion here I realize. So this udpdriver was never meant to be a "sim_cf driver". Maybe @ataffanel  can give us a bit of history.

It seems sim_cf has there own protocol over udp: https://github.com/wuwushrek/sim_cf/blob/master/crazyflie_gazebo/src/gazebo_cfHandler_plugin.cpp that is used to  interface with the simulated cv.

I am sure our udpdriver could be made to work with it in some ways (a tip is that `1, 82` probably should be interpreted as an `ack` packet (the 1) and 82 wold then be `0x52 => port 5 channel 2` which is _log port_ and _logdata channel_.

You might be able to turn the udpdriver into something that works, and we could consider renaming it to simcfdriver.py or similar .)

poritech, 2021-06-14:
Thank you!
I'm not familiar with Crazyflie's communication protocol, need to learn about it

jonasdn, 2021-06-14:
I am closing this issue, feel free to open a new if you have more questions. Or, you could ask in the forum (https://forum.bitcraze.io/) where more users can answer, or at our Discord server (https://discord.gg/dJMtnPZnvU).

To learn about the communication protocol (CRTP) you can check out https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/crtp/ it might not be totally up-to-date, so you should also check out the code at https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/interface/crtp.h or in the python lib at https://github.com/bitcraze/crazyflie-lib-python/blob/master/cflib/crtp/crtpstack.py

Good luck!

PR #244 (2021-06-14): Update udpdriver
- udpdriver: Handle socket import better
- udpdriver: Parse uri instead of hard coding
- udpdriver: Encode bytes before sending
- udpdriver: Make sure we interpret data as uint

Fixes: #240
closed 2021-06-14 by jonasdn