bitcraze/crazyflie-lib-python #257: Add P2P mode with relaxed CTRP timing

ataffanel, 2021-09-09:
When connected to a Crazyflie, the lib will communicate as fast as possible with the Crazyflies. This means that there is no radio time left when using the P2P functionality implemented in bitcraze/crazyflie2-nrf-firmware#29 which prevents Crazyflie to talk to each-other the lib/client is connected to a Crazyflie.

One stop-gap solution, until we improve the radio link further, is to reduce the communication rate of the lib to free radio time for the P2P communication.

This could be done by adding one more optional parameter to the link: `radio://0/80/2M/E7E7E7e7E7?reduced_rate=10`. There is still at least 2 undecided questions:
 - What should the parameter name be?
 - Should the amount of bandwidth reduction be configurable?

whoenig, 2021-09-09:
Would that simply reduce the ping-packets, or all packets? In the latter case, we would like need to implement it as a sleep in the radio thread between USB requests. If that's the case, we should make the sleep time configurable (and default to 0 for highest performance).

ataffanel, 2021-09-13:
Yes, this is exactly what I am envisioning, optionally adding a sleep between USB/radio transfers. Any clue about a good name for it?

whoenig, 2021-09-13:
what about `rate_limit`? This is better than `reduced_rate`, because it indicates that it is an upper limit, not the exact rate. The unit should probably be Hz.

ataffanel, 2021-09-14:
> what about `rate_limit`? 

Sounds good! I pushed the change in a PR.

PR #259 (2021-12-23): Add rate limiter to Radio link
This will help with P2P connection by making the radio leave some air-time for P2P packets.

A new query "rate_limit=" is added to the radio URI. For example to limit the radio polling rate to 100Hz the uri would be: `radio://0/80/2M?rate_limit=100`

Fixes #257
commit ee71aed6 (2021-12-23): Decode radio uri with urllib and add rate_limit

Fixes #257
closed 2021-12-23 by jonasdn