bitcraze/crazyflie-lib-python #258: How To Get Instant Log During Flight

alperenoztas, 2021-09-11:
Hello,

I am working on virtual-reality integration with crazyflies.  I've made the user interface with few commands such as take-off,land and formations.  However, I'd like to see current x-y-z-yaw values on my interface too. I checked the basiclog.py but it didn't work for me. How can I get these parameters instantly during flight?

ataffanel, 2021-09-13:
Hi, You will have to use the logging subsystem so basiclog.py is typically a good starting point. What did not work for you?

alperenoztas, 2021-09-13:
I got kalman x-y-z datas by creating LogConfig instance. But, I'd like to know in what range period_in_ms(DEFAULT = 500ms), when we create our object, works fine. I've observed it in 100ms but I got weird behaviour. I want to know the ideal value of period_in_ms  in order to fly multiple crazyflies synchronously. Thanks in advance.

ataffanel, 2021-09-14:
Could you define what you mean by "weird behavior"? How many drone are you connecting at the same time and what is your expectation in terms of update rate?

A good rule of thumb is that one radio can handle at least 500 packets per seconds in each direction and each log block uses one packet per log. So it should be possible to log at 100Hz a couple of log blocks. Then, adding radios, will help increasing the bandwidth when connecting to more Crazyflies.

alperenoztas, 2021-09-14:
We had tested it with only one crazyflie (We will test it later with 24 crazyflies). When we change it to 100 packets per seconds it doesn't follow all waypoints (We've tried for rotating due to imaginary point but only followed 5/18 waypoints ). However there was no problem with 500 packets per second.

closed 2021-09-15 by alperenoztas