bitcraze/crazyflie-lib-python #260: param.get_value() fails if called "too soon" upon connection

mbaytas, 2021-09-18:
Calling `param.get_value()` immediately upon connection returns `None`:

    with SyncCrazyflie(cf_uri, cf=cf) as scf:
        n_effects = cf.param.get_value('ring.neffect')
        print(n_effects)
    
    >>> None

Adding a delay resolves the issue:

    with SyncCrazyflie(cf_uri, cf=cf) as scf:
        time.sleep(1)
        n_effects = cf.param.get_value('ring.neffect')
        print(n_effects)
    
    >>> 16

jonasdn, 2021-09-23:
Hi!

Yes, this is true, it is somewhat documented in the docstring:

```
    def get_value(self, complete_name):
        """
        Read a value for the supplied parameter. If None then the value has
        not been updated yet, retry.
        """
        try:
            [group, name] = complete_name.split('.')
            return self.values[group][name]
        except KeyError:
            return None
```

This is because the Crazyflie class, today, does not wait until it has received all values before saying that it is  connected. We have tried to change the implementation<sup>[1](#myfootnote1)</sup> to wait for all values to be received. It will take longer time to reach connected, that could be ok. But it also broke<sup>[2](#myfootnote2)</sup> our tests because some code rely on update callbacks being run even if you set up the callback after the CF is connected. So we ended up reverting<sup>[2](#myfootnote2)</sup> it.

So, the next step is a bit unclear, we might try again with the wait. Or we could add an optional wait/timeout to the get_value call, or an optional parameter to the `open_link()` call or the `SyncCrazyflie init()`.

Any thoughts? @mbaytas? @ataffanel?

<a name="myfootnote1">1</a>: #248
<a name="myfootnote2">2</a>: https://github.com/bitcraze/crazyflie-testing/actions/runs/1152680061
<a name="myfootnote1">1</a>: #252

dominiknatter, 2021-09-27:
Hi guys,
today I was testing keyboard-based flights using your script [multiranger_pointcloud.py](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/multiranger/multiranger_pointcloud.py). I tried to adapt the flight speed by setting the parameter posCtlPid.rpLimit via param.set_value(). However, _sometimes_ the code outputted an error "Variable posCtlPid.rpLimit not in TOC".

Reading this post I wonder whether my error is also related to the fact that the internal functionality did not finish something before set_value() was called? if this is the case, then also for set_value() it should be "somewhat documented in the docstring", right?

mbaytas, 2021-09-27:
I would imagine that adding a delay to `get_value()` may be the less preferable option. There are situations where we may call `get_value()` during flight, and any noticeable delay would be undesirable.

ataffanel, 2021-10-04:
What about adding some kind of semaphore to the TOCs to mark if the fetching is completed or not: all function that needs to access the TOC or the param values will have to wait on the corresponding semaphore. This will not slow down the access after the connection is fully complete, but will lazily block thread that request the resource too early.

In python it seems that the `Event` type is what we need here: https://docs.python.org/3/library/threading.html#event-objects.

jonasdn, 2021-10-04:
> What about adding some kind of semaphore to the TOCs to mark if the fetching is completed or not: all function that needs to access the TOC or the param values will have to wait on the corresponding semaphore. This will not slow down the access after the connection is fully complete, but will lazily block thread that request the resource too early.
> 
> In python it seems that the `Event` object is what we need here: https://docs.python.org/3/library/threading.html#event-objects.

I like this. How about we add tests for these things to https://github.com/bitcraze/crazyflie-testing?

We have `get_value()` from this issue, can you think of other functions? And also: could this cause deadlook in some cases?

jonasdn, 2021-10-04:
I updated the test: https://github.com/bitcraze/crazyflie-testing/commit/2f71ab030af2a12680ec68fd7b25e06788a5c7ff

jonasdn, 2021-10-05:
Now the `get_value()` call should block for up to 60s (and then throw exception) waiting for values to be fetched. This blocking will only be for the first time after a connect if it called "too soon".

PR #266 (2021-10-05): param: Add _initialized semaphore
This semaphore will be `set()` when all values have been updated at
least once. This will allow us to wait in get_value to make sure we
have values.

Fixes: #260
closed 2021-10-05 by jonasdn