bitcraze/crazyflie-lib-python #261: param.get_value('ring.neffect

mbaytas, 2021-09-18:


mbaytas, 2021-09-18:
n_effects = cf.param.get_value('ring.neffect')

    print(type(n_effects))
    >>> <class 'str'>
    print(n_effects)
    >>> 16

The correct output should be:

    print(type(n_effects))
    >>> <class 'int'>
    print(n_effects)
    >>> 13

ataffanel, 2021-09-20:
Why do you expect `13` effects? The latest firmware has 18 led ring effects available, so the value returned for `ring.neffect` is correct: https://github.com/bitcraze/crazyflie-firmware/blob/c1150b8039ca5b4996ca4a33831bb15be8939332/src/deck/drivers/src/ledring12.c#L950-L968.

mbaytas, 2021-09-21:
I was expecting 13 effects because:

1. The list of effects in [the docs](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/api/params/#ring) shows 13 of them (screenshot below).
2.  I remember getting some sort of error when I called `cf.param.set_value('ring.effect', 13)`. However, I'm unable to reproduce it now.
 
In fact, testing all effects with an up-to-date and properly assembled Crazyflie right now, I confirm that 18 effects are working. My bad.

However, in this case, the docs for the `ring.effect' parameter may need a little update, see below.

[image]

closed 2021-09-21 by mbaytas

knmcguire, 2021-09-22:
I created this PR to add the missing modes to the documentation: https://github.com/bitcraze/crazyflie-firmware/pull/851

Let me know if that looks good

jonasdn, 2021-09-22:
Fixed by merging fix of documentation in crazyflie-firmware: [https://github.com/bitcraze/crazyflie-firmware/commit/1b7ed621df3d68ad7445ffc09d5b052479a139ed](url)