bitcraze/crazyflie-lib-python #263: Variable ctrltarget.z is not in TOC

ayush7138, 2021-09-22:
I'm trying to tune the PID gain for my crazyflie, when trying to perform a step response using the script i get this error. I'm not familiar with what TOC is or how to solve this. 
Here is the error I got: 
Connecting to radio://0/80/2M/E7E7E7E723
Connected to radio://0/80/2M/E7E7E7E723
Waiting for estimator to find position...
update info
Sending to the zposition PID controller: Kp: 5.1, Ki: 0.5, Kd: 0.5
update info
do step
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python3.6/tkinter/__init__.py", line 1705, in call
    return self.func(*args)
  File "PID_controller_tuner.py", line 224, in do_step
    with SyncLogger(self.cf, log_config) as logger:
  File "/home/baza/crazyflie/crazyflie-lib-python/cflib/crazyflie/syncLogger.py", line 108, in enter
    self.connect()
  File "/home/baza/crazyflie/crazyflie-lib-python/cflib/crazyflie/syncLogger.py", line 67, in connect
    self._cf.log.add_config(config)
  File "/home/baza/crazyflie/crazyflie-lib-python/cflib/crazyflie/log.py", line 475, in add_config
    raise KeyError('Variable {} not in TOC'.format(var.name))
KeyError: 'Variable ctrltarget.z not in TOC'

jonasdn, 2021-09-23:
Hi, thanks for filing an issue!

This seems a bit strantge, I will try to reproduce it later today. To help out, could you perhaps modify the `basiclog.py` example to add the variable `ctrltarget.z` and run it to make sure that it gets printed?

Jonas

dominiknatter, 2021-09-27:
could this in any way be related to [this issue](https://github.com/bitcraze/crazyflie-lib-python/issues/260)? Just struck me, but maybe I'm wrong

knmcguire, 2022-03-07:
I tried out the logging, and this issue seems to be solved after this fix in the firmware : https://github.com/bitcraze/crazyflie-lib-python/pull/266 

Closing the issue!

closed 2022-03-07 by knmcguire