bitcraze/crazyflie-lib-python #301: Implement notifySetpointsStop packet

knmcguire, 2022-01-14:
When switching from position setpoints to high level commander set points, it is necessary to implement the notifySetpointStop packet in the cflib api. 

see https://github.com/bitcraze/crazyflie-firmware/pull/903

qubitter, 2022-03-15:
Maybe I don't understand, but what's the difference between notifySendpointStop and HighLevelCommander.stop()?

Doesn't really seem like we're doing much with setpoint in HighLevelCommander.

knmcguire, 2022-03-15:
Hi! This is a functionality that exist in Crazyswarm and the firmware but not in the cflib. It is meant for better switching between regular setpoints and HL commands. So yes you are right, this is not a High level commander functionality.

PR #383 (2023-02-20): Add notify setpoint
This adds the notify_setpoint_stop to the cflib 

I still need to flight test this, and we need to know if we bump the crtp version as well? 

This should be decoded by this part of the crazyflie-firmware
https://github.com/bitcraze/crazyflie-firmware/blob/674653717bede0f0d205a0e112585aa210109d3f/src/modules/src/crtp_commander.c#L94-L103

closes #301
closed 2023-02-20 by knmcguire