bitcraze/crazyflie-lib-python #310: PositionHLCommander activates PID controller by default.

jonasdn, 2022-01-25:
Let's switch to not actiavte any controller by default and have example scripts set PID controller.

knmcguire, 2022-01-25:
I'm maybe a bit more curious about the thought behind here. I actually think that the PID should be default always, as mellinger and INDI are still considered for development only and the PID controlller is the most robust. There has been to many times that users have been confused because they used a script that enabled mellinger, see if flip because they use an imperfect positioning system like Loco or flowdeck and make wrong assumptions. 

But in a sense, as long as it is default in the firmware, it would not be necessary to specifically specify this in the scripts indeed. Perhaps we should also try to find scripts that still enable mellinger and comment out those lines

jonasdn, 2022-01-25:
> I'm maybe a bit more curious about the thought behind here. I actually think that the PID should be default always, as mellinger and INDI are still considered for development only and the PID [...]

Yes, the underlying issue here was that since the client used the PositionHLCommander the PID controller was selected on each client connect. And it might be a bit surprising that just instanciating the class will switch the controller. Now we set PID controller in all examples. But do not set a controller underneath the hood, so to speak.

commit 06c237dc (2022-01-25): PositionHLCommander: Do not set controller by default

Closes: #310
PR #311 (2022-01-25): PositionHLCommander: Do not set controller by default
Closes: #310
closed 2022-01-25 by jonasdn