bitcraze/crazyflie-lib-python #322: Ready To Fly Mode

Moj-Dev, 2022-03-04:
I'd like to define a ready to fly mode in my application that motors spins in lowest rpm before takeoff. Could you give me help that what is the best way to implement this  using python Crazyflie library?

Moj-Dev, 2022-03-04:
I implemented it using python library. Here is the code you can use:
        roll = 0
        pitch = 0
        yawrate = 0
        print("let's go")
        thrust_mult = 1
        thrust_step = 500
        thrust = 20000
        cf.commander.send_setpoint(0, 0, 0, 0)
        for r in range(10):
            cf.commander.send_setpoint(roll, pitch, yawrate, thrust)
            time.sleep(0.1)
        print("Ready 2 fly")

knmcguire, 2022-03-07:
This is a duplicate of the issue in the crazyflie firmware: https://github.com/bitcraze/crazyflie-firmware/issues/978

Let's discuss there if this needs to be in the cflib or not, but for now I'll close this one.

closed 2022-03-07 by knmcguire