bitcraze/crazyflie-lib-python #331: Inaccurate trajectory when using 0.1.18 compared to 0.1.16

zp-yang, 2022-05-09:
Hi, we were using 9 crazyflie 2.1 equipped with active marker decks and Qualisys mocap system to fly them in accurate formation in our lab. The old cflib(0.1.16) gave us better trajectory compared to the newest cflib(0.1.18)
[image]
[image]
[image]
[image]
[image]
[image]

We tested with 6 crazyflies out of the 9 in our formation. Here is the link to the measurements captured on QTM.
https://drive.google.com/drive/folders/1x4dFj9PGpjKA3M_sDSvTToho24CW3U_u?usp=sharing

Here is the flight script and trajectory data we used for both tests.
https://github.com/jgoppert/pybswarm/blob/master/scripts/flight_qualysis.py
https://github.com/jgoppert/pybswarm/blob/master/scripts/data/geometry_9drone.json

Any idea why this is happening? Please let us know how we can help to resolve this issue, thanks!

knmcguire, 2022-05-12:
This is indeed very strange... I can't think of any addition we did that could have caused this.

 Could you let us know which version of the crazyflie-firmware you have flashed on the Crazyflie? Is that constant on both libraries?

zp-yang, 2022-05-12:
The firmware version was the same on both tests. We recorded these two sessions back-to-back and only switched the cflib version. However, I do not remember the exact firmware version on our drones. Is there a way to check the current firmware version?

knmcguire, 2022-05-13:
You can check the console tab in the cfclient or the console.py script of the cflib examples when connecting with the Crazyflie

zp-yang, 2022-05-13:
Looks like the crazyflie firmware is on `2022.03` and NRF51 firmware is on `2022.0`

Here is the output from console.py, hope this information is useful.
```
SYS: ----------------------------
SYS: Crazyflie 2.1 is up and running!
SYS: Build 0:4d476a20f811 (2022.03) CLEAN
SYS: I am 0x203934305333500800390035 and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_CORE: 2 deck(s) found
DECK_CORE: Calling INIT on driver bcActiveM for deck 0
ACTIVE_MARKER: Deck FW Qualisys1.0
DECK_CORE: Calling INIT on driver bcLedRing for deck 1
IMU: BMI088: Using I2C interface.
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Complementary (1) estimator
CONTROLLER: Using PID (1) controller
MTR-DRV: Using brushed motor driver
SYS: About to run tests in system.c.
SYS: NRF51 version: 2022.0
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
DECK_CORE: Deck 0 test [OK].
SYS: Self test passed!
STAB: Wait for sensor calibration...
SYS: Free heap: 19696 bytes
STAB: Ready to fly.
[host] Connected, use ctrl-c to quit.
```

knmcguire, 2022-05-16:
Thanks for the info! I'm not sure if updating the firmware will help in this case though, but it would be good to try out. We also have a new release of the cflib as well (0.1.19)

It seems that the controller is sluggish in the second plot with 0.1.18... but it is difficult to imagine that this is due to the cflib except for if there is a big delay with sending the control commands. It could be that in the first lib, the mellinger controller is selected and in the second one, the PID controller, but if the scripts are exactly the same I would find it unlikely that the controller would suddenly switch like that... just thinking out loud here.

I've been looking at the release notes of [0.1.17](https://github.com/bitcraze/crazyflie-lib-python/releases/edit/0.1.17), [0.1.17.1](https://github.com/bitcraze/crazyflie-lib-python/releases/tag/0.1.17) [0.1.18](https://github.com/bitcraze/crazyflie-lib-python/releases/tag/0.1.18) and there might be some intersting addition that might have affected this:
- https://github.com/bitcraze/crazyflie-lib-python/pull/299
- https://github.com/bitcraze/crazyflie-lib-python/pull/312
- in https://github.com/bitcraze/crazyflie-lib-python/pull/311

Also, [persistent parameters](https://www.bitcraze.io/2021/12/powerful-persistent-parameters/) is a new addition in between the release of 0.1.16 and 0.1.17. Not sure if this could happen, but maybe there is some problem with that somehow, that it sets a different controller as persistent? Perhaps that is something to double check with the cfclient.

Btw, on which OS are you running this? Windows or Ubuntu?

zp-yang, 2022-05-16:
The flight script was running on Ubuntu 20.04 and we were using 3 crazyradios on a usb hub with each radio connecting to 3 drones. We will try updating the firmware to the latest and let you know if that helps. Thanks!

zp-yang, 2022-05-20:
My colleague tried another run with the latest crazyflie firmware `2022.05` and cflib `0.1.19` but unfortunately the problem still persists.

krichardsson, 2022-05-23:
I'm not sure what might be causing this,
#311 should not change anything as far as I can see in your case
#299 and #312 are related to USB connections, I can not see that it should change anything in your case unless libusb-package has some side effect on the radio connection that modifies the timing, but it sounds unlikely.
It is a bit confusing.

I can also see that we made a change to the QTM example, as far as I can remember this was due to a change in the QTM lib. Is it possible that the differences you see are related to QTM somehow? For instance if you have slightly more latency when pushing position information to the Crazyflie could possibly lead to the results you see. I have no idea why this should change between 0.1.16 and 0.1.18 though.
Considering that the high level commander is running in the Crazyflie and that the python lib is not really involved in the actual flying, might also point in the direction of latency in the position data.

Did you also try 0.1.17?

knmcguire, 2023-04-12:
So at our last triage meeting, we decided to close this one as it has been inactive for a year, but if this is still a problem please reopen

closed 2023-04-12 by knmcguire