bitcraze/crazyflie-lib-python #340: Using URI with address doesn't work with cfloader

evoggy, 2022-06-03:
It's been reported ([here](https://github.com/orgs/bitcraze/discussions/57) and [here](https://github.com/orgs/bitcraze/discussions/68#discussioncomment-2873496)) that using an URI with an address will fail the flashing of the AI deck.

Although the cfloading is not part of this repository, my guess is that this issue is in the library somewhere.

knmcguire, 2023-04-12:
some thoughts on this!
- The uri handling in the cflib is not generic enough
- cfloader is a bit of a mess and at one point should be rewritten

I haven't seen this lately.. but it miht be related to this: https://github.com/bitcraze/crazyflie-clients-python/issues/616

ataffanel, 2025-02-06:
This has already been fixed, likely when working with the new bluetooth stack.

closed 2025-02-06 by ataffanel