bitcraze/crazyflie-lib-python #346: Crazyflie's are falling when they are top of each other

alperenoztas, 2022-07-05:
Hello There!

We are trying to make a 3D shaped formations with more than 8 crazyflies. However, when they come top of each other crazyflies , they are starting weird behaviour and immediately falls. What can we do to fix this problem? Also, can we change controller type in motion commander just like in high level commander ? (PID to Mellinger)

whoenig, 2022-07-06:
When two or more UAVs are on top, there is an aerodynamic effect known as downwash. The easiest/best way to deal with the issue is to avoid these configurations during planning.

Yes, you can change the controller in any flight mode and at any time (even mid-flight).

alperenoztas, 2022-07-06:
Thanks for the quick reply. We will make some changes on our configuration and will test Mellinger during flight.