bitcraze/crazyflie-lib-python #351: Getting Fatal Python error while running ramp.py example.

salocinx, 2022-08-04:
Hi. I try to run [ramp.py](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/motors/ramp.py) on Windows 10 with Python 3.10.6 64bit installed. The rotors ramp up, but when the process finishes, I get the following error:

```
PS E:\crazyflie\tests> python .\ramp.py

Connecting to radio://0/10/2M/E7E7E7E7E7
Disconnected from radio://0/10/2M/E7E7E7E7E7
Fatal Python error: _enter_buffered_busy: could not acquire lock for <_io.BufferedWriter name='<stdout>'> at interpreter shutdown, possibly due to daemon threads
Python runtime state: finalizing (tstate=0x0000028f52c7bad0)

Current thread 0x00002568 (most recent call first):
  <no Python frame>
```

Am I doing something wrong?

knmcguire, 2022-08-05:
Hi!

Yes I'm able to recreate this with windows 11 and python 3.10. 

It is due to this line somehow:

https://github.com/bitcraze/crazyflie-lib-python/blob/8b6f740eb131083d024f9129073b9b8bf0f735f2/examples/motors/ramp.py#L65

But on the first look there doesn't seem there is anything wrong with it? There is probably some bad thread handling going on here but we will need to investigate. 

Thanks for reporting!

salocinx, 2022-08-06:
Hi! The example basically runs fine and the issue only arises shortly before the process finishes. No hurry at all. Thanks for keeping an eye on this.

knmcguire, 2022-08-08:
ah oke! For me it actually fails immediately and the motors don't spin up at all.

closed 2025-10-07 by ataffanel