bitcraze/crazyflie-lib-python #361: Make trajectory uploader part of the cflib

knmcguire, 2022-09-18:
Currently the high level commander trajectory is partly integrated in the cflib ([define_trajectory](https://github.com/bitcraze/crazyflie-lib-python/blob/8d82a769952d1b49de0d6779a2b59ba94036151a/cflib/crazyflie/high_level_commander.py#L180)) and partly defined in examples ([uploader class in autonomous_sequence_high_level.py](https://github.com/bitcraze/crazyflie-lib-python/blob/8d82a769952d1b49de0d6779a2b59ba94036151a/examples/autonomy/autonomous_sequence_high_level.py#L65-L87)). 

It would be great if the trajectory upload class would be part of the cflib that the user only has to call 'upload'

knmcguire, 2022-12-02:
There is talk about that the cflib is getting too bulky and that perhaps this should not be part of it. However, @ataffanel argued the trajectory functionality is a functionality that exists within the Crazyflie firmware that usually should be supported fully from the cflib side (even more so than the positioning support).

whoenig, 2022-12-02:
I think this should certainly be part of it - all features of the firmware should be usable with cflib. You don't have to include example trajectories, but it should be possible to upload a 2D array with the coefficients (e.g., that one can load from a csv file like done in the Crazyswarm).

closed 2022-12-07 by krichardsson