bitcraze/crazyflie-lib-python #363: multi-threading issue in get_estimated_positions

orimana2020, 2022-11-12:
## Description
I am trying to call get_estimated_position in a loop and it stuck after 254 iterations and when I exit the program I got this error.

## error:
`Traceback (most recent call last):
  File "/home/ori/crazyflie-lib-python/examples/swarm/hl-commander-swarm.py", line 95, in <module>
    print(swarm.get_estimated_positions())
  File "/home/ori/crazyflie-lib-python/cflib/crazyflie/swarm.py", line 148, in get_estimated_positions
    self.parallel_safe(self.__get_estimated_position)
  File "/home/ori/crazyflie-lib-python/cflib/crazyflie/swarm.py", line 278, in parallel_safe
    thread.join()
  File "/usr/lib/python3.8/threading.py", line 1011, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.8/threading.py", line 1027, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt`

## Code snippet:
 
    import time

    import cflib.crtp
    from cflib.crazyflie.swarm import CachedCfFactory
    from cflib.crazyflie.swarm import Swarm

    def activate_high_level_commander(scf):
        scf.cf.param.set_value('commander.enHighLevel', '1')

    def activate_mellinger_controller(scf, use_mellinger):
        controller = 1
        if use_mellinger:
            controller = 2
        scf.cf.param.set_value('stabilizer.controller', controller)

    def run_shared_sequence(scf):
        activate_mellinger_controller(scf, False)

        box_size = 1
        flight_time = 2

        commander = scf.cf.high_level_commander

        commander.takeoff(1.0, 2.0)
        time.sleep(3)

        commander.go_to(box_size, 0, 0, 0, flight_time, relative=True)
        time.sleep(flight_time)

        commander.go_to(0, box_size, 0, 0, flight_time, relative=True)
        time.sleep(flight_time)

        commander.go_to(-box_size, 0, 0, 0, flight_time, relative=True)
        time.sleep(flight_time)

        commander.go_to(0, -box_size, 0, 0, flight_time, relative=True)
        time.sleep(flight_time)

        commander.land(0.0, 2.0)
        time.sleep(2)

        commander.stop()

    uris = {
        'radio://0/80/2M/E7E7E7E7E7',
        # 'radio://0/30/2M/E7E7E7E712',
        # Add more URIs if you want more copters in the swarm
    }

    if __name__ == '__main__':
        cflib.crtp.init_drivers()
        factory = CachedCfFactory(rw_cache='./cache')
        with Swarm(uris, factory=factory) as swarm:
            swarm.parallel_safe(activate_high_level_commander)
            swarm.reset_estimators()
            # swarm.parallel_safe(run_shared_sequence)
            for i in range(1000):
                print(i)
                print(swarm.get_estimated_positions())

krichardsson, 2022-11-15:
Hi!

This sounds like there might be a bug, possibly in the firmware (254 sounds like a uint8).

Even though there might be a bug, the way you use this function is not recommended and I suggest you find a different way of getting the position. The documentation should be improved to point out that it is very costly in resources to use the function. 

The flow of events is (for each call on each Crazyflie):
1. Set up a log block for the position log
2. Wait until one log of the position arrives
3. Remove the log block again

The proper way would be to set up the log block once and start the logging. When you need to use the position, use the latest position value that has been received. 

You can perhaps use the [start_position_printing()](https://github.com/bitcraze/crazyflie-lib-python/blob/8d82a769952d1b49de0d6779a2b59ba94036151a/examples/autonomy/autonomousSequence.py#L116-L124) for inspiration?

ArisMorgens, 2025-02-13:
Fixed by adding warning in #513.

closed 2025-02-13 by ArisMorgens