bitcraze/crazyflie-lib-python #407: Velocity and acceleration controllers for crazyflie with motion capture system.

gaurav713914, 2023-06-28:
Is there any way through which we can control Crazyflie's velocity or acceleration while using the Qualisys motion capture system for feedback? When we use a flow deck, we already have a motion commander for velocity control. I want to use the velocity command with the Qualisys motion capture system. Please help regarding the same.

krichardsson, 2023-06-29:
When using an external system for positioning (like Qualisys) the position is sent to the Crazyflie via the radio. If you are using the python lib, you should use the [ExtPos class](https://www.bitcraze.io/documentation/repository/crazyflie-lib-python/master/api/cflib/crazyflie/extpos/) 
The position information is pushed into the Kalman estimator and is fused with any other sensor information that is available, the same way as the flow information is. 
As the Motion commander is only related to sending setpoints to the controller in the Crazyflie, you can use it the same way regardless of what information you feed to the estimator. 

You can find an example of how to get data from a Qualisys system and feed it to the CF in [qualisys_hl_commander.py](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/qualisys/qualisys_hl_commander.py), it uses the high level commander instead of the Motion Commander though.

gaurav713914, 2023-06-29:
Thank you for the clarification, I tried it and I was able to run the motion commander but I faced one problem during feedback that the 6d data which QTM is publishing in python has coordinates (x,y,z) and a rotation matrix. We need roll, pitch and yaw separately, also it will be good if we can receive velocity and acceleration data of any rigid body from QTM. Can you please help regarding it?

krichardsson, 2023-06-30:
> We need roll, pitch and yaw

[Scipy has nice support for converting between various flavors of rotation representations.](https://docs.scipy.org/doc/scipy/reference/generated/scipy.spatial.transform.Rotation.html#scipy.spatial.transform.Rotation)

> it will be good if we can receive velocity and acceleration data of any rigid body from QTM. Can you please help regarding it?

I'm not that good at QTM, [Qualisys](https://www.qualisys.com/) will be a better source of information.

gaurav713914, 2023-07-08:
Thank you @krichardsson for support on roll, pitch and yaw. It worked as per your suggestion.

closed 2023-07-08 by gaurav713914